bigtreetech/BIGTREETECH-TFT35-V3.0 #2: [QUESTION] Source files

Patag, 2019-08-24:
Do you intend to publish source files allowing customisation ?

kordris, 2019-08-25:
I would like to know this also. because as it is now.. I can't use the touch screen part on my flyingbear tornado because it has two extruders. I use the second extruder mostly. So... I can't setup the screen for using two extruders. I have look everywhere for a update or something to update this screen so I can enable the second nozzle.

thisiskeithb, 2019-09-03:
Or @Msq001? It would be really nice to be able to fix/customize this firmware.

Mistergrow, 2019-09-05:
@bigtreetech Hello?Answer please

spilz87, 2019-09-07:
@bigtreetech Hello?Answer please

bigtreetech, 2019-09-07:
will released next week. maybe Tuesday or Wednesday

spilz87, 2019-09-07:
Very good news :)

Thanks

Baltazart, 2019-09-07:
Un grand Merci.......

Aleks-MO, 2019-09-08:
@Bigtreetech Will WiFi work on TFT35v3.0 when?

bigtreetech, 2019-09-10:
@Aleks-MO we will implement it this week. ^_^

p1ngb4ck, 2019-09-11:
@bigtreetech : got some (bad) experiences with the old module-type on a different printer (MKS-Gen-L, MKS TFT v4.0 + MKS WiFi). The idea behind that setup is basically the same. Can you maybe answer some questions I got towards the wifi function in general / confirm my assumptions ?
- Wifi is realized via "typical" wifi to serial connection I assume ?! - so probably available in tft serial mode only? Or will the serial connection from the wifi to the skr (1.3 in my case) will work in 12864 mode too, as long as the serial cable is connected?
- I managed to get connection via tcp <ip-address>:<port> in pronterface and use single gcodes (e.g. start abl) - but actually never achieved to print via wifi - it always stopped after few lines of gcode of the file I wanted to print - do you have the option to test the wifi-connection with repetierserver ? I`d offer to do betatesting if needed, I *think* I got the necessary background for that.
- can you maybe give me a heads up about the differences between ESP-01 and ESP-01S ? I mean technically? TY :)  :+1:

spilz87, 2019-09-11:
Sources files shared !!! :)
Thanks @bigtreetech

Patag, 2019-09-11:
Where ? Can't find them

extesy, 2019-09-11:
@Patag https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

Patag, 2019-09-11:
Thanks @extesy 
Can I close this one ?

Patag, 2019-09-11:
I hope @spilz87 will open a new topic with his very pertinent questions

closed 2019-09-11 by Patag

spilz87, 2019-09-11:
@Patag you will find my new topics in firmware repository as I worked on TFT24 V1.1 ^^
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues