bigtreetech/BIGTREETECH-TFT35-V3.0 #3: Installing ESP-01S

rlvanhuis, 2019-09-04:
What is the correct pinout for the ESP-01 module?  It appears the ESP module needs to be inverted to apply power correctly.  The other pins do not match the locations either.  Just trying not to let the smoke out.  Very little mention in the manual.  Is programming needed?

Mistergrow, 2019-09-05:
Have the same Problem, it would be nice when @bigtreetech will respond.

Mistergrow, 2019-09-06:
I wrote a Mail and they told me Wifi is not realized yet. The engineers are ready and we should look forward.

p1ngb4ck, 2019-09-11:
please be sure to notice + care for the difference between ESP-01 and ESP-01S (notice the S!) - it`s different versions. The TFT35 3.0 uses the ESP-01S !

p1ngb4ck, 2019-09-11:
Just a little shoutout to the btt-devs :
My WiFi - adapter arrived yesterday already, today cam the tft35 3.0 - and just few hours before I will have the time to set it all up, you provide the new info + fw for wifi - that`s exactly what I call just in time =)
I like ! Thanks a bunch :D

p1ngb4ck, 2019-09-11:
@Mistergrow could you do a test-print via wifi already?

spilz87, 2019-09-11:
What software do you use on your esp8266 ?

extesy, 2019-09-11:
Where is firmware for ESP-01S (esp3d.bin file)?

Mistergrow, 2019-09-11:
@extesy look here: 

https://github.com/luc-github/ESP3D

Mistergrow, 2019-09-11:
@p1ngb4ck No sorry, I have some issues. I can connect an control the Printer but I cant see files from SD Card on my PC. Now I found out that my Display freeze when I connect a USB Flash Drive. There are two LED´s on this board but only the SD Led is green when a card is inside not the Power Led...Tomorrow I will contact BTT Support and ask for help.

rlvanhuis, 2019-09-11:
Conflicting diagrams?
What version is correct?  I have a V3.0.  Are there two different V3.0 controllers?  Still very concerned about burning out the display.  
Is there a harness missing from the kit?  The new guide seems to indicate another harness is needed.  The picture shown on AliExpress clearly shows the ESP-01S module as the proper device.  I have tried to locate another 8266 with a different pinout, no luck.

Mistergrow, 2019-09-11:
When you compare this pinout: 

https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/blob/master/BIGTREETECH%20TFT35%20V3.0/Hardware/TFT35%20V3.0-PIN.pdf

with the new one: 

https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/blob/master/BIGTREETECH%20TFT35%20V3.0/Hardware/TFT3.5%20V3.0-WIFI%20User%20Guide.pdf

You see it is different. The Wifi Manual has two Pictures of ESP-S01 Top and Bottom(with Pinout). 

"_ESP using Wifi module and you can directly insert the module..._" 
The Picture shows the Top side of my ESP8266
Then the Wifi Manual shows the bottom side of the ESP-8266(with Pinout):
_"Overlay of ESP8266 using Wifi Module use the Dupond Line to connect to the Pin"_
Maybe the ESP32 has a matching Pinout and you can connect it directly.  
I connect dupond cables and Wifi is working, except the SD&USB probs I have (with and without Wifi module) 
Maybe the ESP32 Pinout is matching with the TFT35V3.0 Wifi Pinout and you can plug it in directly...?

@bigtreetech why picture of same Module (ESP8266 ESP015 S Series) Top and bottom Pictures with different description for ESP/ESP8266 and ESP32(top of manual) with different connection methods?

Mistergrow, 2019-09-13:
> @Mistergrow - yes you can plug it directly... Like it is also stated in the manual ;D
> Did not achieve to get esp3d on my board to test, yet though :(

Are you sure?When I Plug in and turn on the Blue Led doesn‘t Flash. And also the Pinout from Display is wrong, RX and TX Pins from diaplay and esp. Tomorrow I will check this again.

p1ngb4ck, 2019-09-13:
Power the esp with just connecting 3.3V + GND in original shipped version - esp-wifi-net is visible, esp01 is booted - no blue led.
I don`t think the blue LED is for "normal" power status, but rather some serial activity LED, that will flash as soon as there is data being transmitted.
Yep, I was wondering about Tx+Rx too - but think about the following : the transmit goes to receive on the other chip and vice versa - I think it is meant to be directly pluggable..

spilz87, 2019-09-13:
The goal of adding ESP8266 to the screen is to add a serial communication with the printer  through WiFi. Do I understand well ?

There is no way to use WiFi through ESP8266 to copy a file on sdcard and be able to run a print from the SD card ?

Thanks

Mistergrow, 2019-09-13:
In E3D you have a function to browse and Card files and Upload. It wasn‘t possible for me to see or upload files. When I refresh screen is blank, when I try to Upload is says something like „only 8.3 filenames allowed“ tomorrow I have time to See whats wrong

spilz87, 2019-09-13:
Ho I see, I guess it’s because marlin only accept 8.3 format file in G-code to run a print.

I didn’t really check what touchscreen really do with esp8266, it’s just a bridge with serial port or it does something else ?

p1ngb4ck, 2019-09-13:
@spilz87 the esp itself is a full stacked embedded device with its own processor and firmware.
it`s not just a wifi bridge, but a mini device connected to wifi and to the printer via serial conn. - providing extra features in addition to the "pure" serial-bridge.
the tft that hosts the esp provides functions for the esp to use the sd-card-slot and the usb via serial - thats why basically all those shiny features like upload to sd etc are possible..

spilz87, 2019-09-13:
Ok i see, I just read ESP3D page 

But what I read too is that esp3d can be directly connected to the motherboard, without tft. But I understand that as tft is already connected to serial port on board, we have to connect esp8266 to tft.

p1ngb4ck, 2019-09-13:
absolutely correct - you could add the esp to the board nonetheless - with dupont wires and correct setup - since most boards have multiple serial ports available... but it`s a lot simpler and more useful to attach it to the tft

BobP2020, 2019-09-13:
I have been working with the ESP-01(S) module and the TFT35 V3.0 display unit from BigTreeTech for a while now and have some insight I would like to share. First, I have found the information provided by BTT to be quite confusing..but then again, I am not an expert on these modules...just learning as I go. First thing I would offer from my experience-The ESP-01 and the ESP-01S have the same pinout. The -01 version has 512K of memory while the "S" version has 1M of memory. I have been using the "S" version but I believe the -01 version will work as well if properly configured when compiling the ESP3d code. Next item is that the "original" TFT35 V3.0 boards supplied DO NOT have a WiFi connection this a simple, direct, plugin for the typical ESP-01 module. If BTT has a revised version of the TFT35 V3.0 display it is not clear to me. My advice is do not plug the ESP-01 module into the WiFi connector of the display without FIRST confirming where the power (3.3V) and ground pins are located. Use a multimeter to verify this. My components are NOT a match for direct plugin (YMMV). I used Dupont style jumpers to connect the pins of the ESP-01 module to the WiFi socket of the display board. Note that a proper connection will have the Tx of the ESP-01 connecting to the Rx of the TFT35 and the Rx of the ESP-01 connecting to the Tx of the TFT35. I first identified the Gnd and 3.3V pins on the TFT35 and connected them to the ESP-01. Fortunately the ESP-01(S) units I got from BBT had the pins labeled so this was easy. I then attached additional jumper cables using the Gnd and 3.3V pins as a reference. It works but with some issues. I will continue to check out the various features of web interface. So far I have tested position/motor controls, hotend monitor and control as well as heated bed monitor and control (initially the bed control was intermittent and need additional checking).  The SD file support was not working for me. Notes on compiling the ESP3D firmware (thanks to the developer(s)...I used Arduino IDE version 1.8.9, ESP8266 library version 2.5.0 (newer one could not get to work) and ESP3D version 2.1.0.b35. I assigned 512K to be used for SPIFFS. I set my baud rate to 250000 because I cannot get the TFT35 V3.0 to work correctly at 115200 (I have an issue open with BTT support on this problem). Hope this is of some help to those trying to use the WiFi feature of their TFT35 V3.0 display.

spilz87, 2019-09-13:
To understand why all my question, here is my idea and my goal with this screen with 2 modes : 
- for me Marlin works very well alone : print directly from SD card ( no Buffer overflow, no bug, etc.)
- I still want to run print directly from SD card with marlin, not send all G-code through serial port
- during a print, a custom page on tft will be nice, but as possible in open loop (get data from marlin through serial port, but try to don’t send data from tft.
- concerning WiFi, same goal : several functions but nothing that can affect a print if it freeze

BobP2020, 2019-09-13:
> 
> 
> Ho I see, I guess it’s because marlin only accept 8.3 format file in G-code to run a print.
> 
> I didn’t really check what touchscreen really do with esp8266, it’s just a bridge with serial port or it does something else ?

As far as I recall, Marlin 2.0 will accept long filenames but they may not show all of the characters. My experience with gcode files on an SD card inserted into the TFT35 display required the file extension be ".gcode", when I tried ".gco" the file did not show on the screen. This was true with the .21 FW. I have not rechecked this with the .23 FW.

spilz87, 2019-09-15:
Actually I tried 3 times to run a print from SD card through TFT mode, and all failed :(
The print stop randomly ...
That’s why I want to find a way to still use tft interface but with marlin directly reading SD card. 
Want the same with esp8266

BobP2020, 2019-09-15:
s

> Actually I tried 3 times to run a

If you are trying to print directly from the TFT screen and NOT via WiFi I think I have the solution. What solved the problem that you describe was to increase the baud rate up to 250000. For some reason the default 115200 does not work...print process starts and then "hangs" the TFT35 display requiring a reset. I have informed BTT of this issue but have not received any feedback from them (I did this via their facebook group). Try it. It does require you to recompile the firmware for the control board but there is a menu entry on the TFT35 display that allows easy switching between to two supported baud rates. The baud rates MUST match.

spilz87, 2019-09-15:
@BobP2020 yes i already read your issue about that, and it’s not the only issue : if you do a 4x4 points leveling, the marlin’s reply is bigger than tft receive buffer and the print just continue to wait a “ok” that was lost because buffer over flow ...

I can’t use something else than 115200 because I have several small electronic cards like esp8266 (with personal code inside) which monitor my printers by listening serial port (like send notification when print is finished or when an error code is send) 

That why I still prefer to find a way to have a tft screen as passive as possible. 

(I already open an issue about using SD card slot directly from marlin. If some people are interested by it too, I suggest to continu the discutions on this other topic to don’t change the subject of this one)

carl1961, 2019-09-23:
easiest way to install the ESP-01S, is to carefully remove the black plastic pin holder, I used a thin blade to get it to raise up and then pull it off. take a soldering iron and heat one pin at a time and shove it through to the other side and re-solder
[image]
[image]
[image]

. then put plastic pin holder back on. you can now plug it in.

p1ngb4ck, 2019-09-23:
@carl1961 did you take the position of the on-pcb-wifi-antenna into account?
I suppose you didn`t ... but unfortunately this makes this way of mounting not recommendable ...
I am almost done doing a complete "preddy in repetierserver connected via esp01s - esp3d-tcp/ip" - doc with all that useful info - I am just doing final optimizations right now... Will share that one here and on FB then...

carl1961, 2019-09-23:
Well actually turned that way faces it out away from the motherboard and towards the outwards direction.

p1ngb4ck, 2019-09-25:
Nope, it turns the antenna to the pcb of the display which will make the connection worse. I have now been able to play around with the esp01s a bit too. Here are my findings:

There seem to be two different TFT3.5 v3.0 pcb layouts. I have the one, that has the esp8266 writing on it and is described on the lower picture. The previous writers were all correct : the esp01(s) can NOT be directly inserted. As the wifi manual says. The connector on this layout is basically mirrored, that's why carls method works - though I don't think it is recommendable due to the position of the wifi antenna on the esp01 pcb being located so, that it faces the display pcb directly.
My advice is the following: solder a decent, well isolated cable, that's not too long.
Also: nearly any good how to states, there should be a 100nF capacitor between v+ and GND - just solder one on the upper side directly to the pins. If you use the wiring mentioned in the wifi tutorial here, the EN pin is open and RST goes to the corresponding port. Other manuals state, to connect EN and RST to 3.3V V+ via a 2.2k resistor - to avoid unintended resets.

Myself, I currently reached the following status:
I replaced the eeprom on the esp01s with a W25Q64 64MBit Flash (8MB) and then used an esp01 flasher (the cheap one) and arduino ide to flash esp3d from lucs github repo. I achieved to get the esp01s configured and connected to the TFT3.5, it connects to my wifi and I can access esp3d with the static ip I set.
I also already achieved to get connection to the data port 8888 with print run, and see temperatures and run simple commands.
For everyone here again: the 8.3 gcode version necessity is due to marlin - not esp3d.
My current problems : I do not get the connection via data port "stable" enough to allow a connection and printing from RepetierServer.
My goal is to be able to use wifi connection to connect my printer to RepetierServer running on a raspberry. The raspberry is connected to the wifi ap via GBit LAN. 

I suspect the problem to be one of the following:
Either indirect serial connection from wifi to tft to board - I use an SKR1.3 - will try to connect the Esp to the SKR1.3 and configure it to be a Bluetooth serial in marlin and run with 250000 baud indepently from tft.
Other problem source could be wrong buffer sizes - does anybody maybe have some detailed info for me, on how to set all the related options in marlin for SKR1.3,TFT3.5 and ESP for a tcp to serial data connection work well? I can exclude wifi problems for testing of course - transmitter and receiver located near to each other for testing - as soon as I get it stable, I can go on testing with less perfect wifi signal...

sl1pkn07, 2019-09-28:
> I replaced the eeprom on the esp01s with a W25Q64 64MBit Flash (8MB) 

this one is valid? https://es.aliexpress.com/item/32828166840.html

greetings

p1ngb4ck, 2019-10-04:
> > I replaced the eeprom on the esp01s with a W25Q64 64MBit Flash (8MB)
> 
> this one is valid? https://es.aliexpress.com/item/32828166840.html
> 
> greetings

Hi
Yes, it is - but you don`t need to replace the eeprom necessarily - I just mentioned I did that, to be able to use esp01-flashing over the air...

sl1pkn07, 2019-10-04:
i want for the same (aunto-update from webUI)

hapklaar, 2019-10-25:
> My current problems : I do not get the connection via data port "stable" enough to allow a connection and printing from RepetierServer.
> My goal is to be able to use wifi connection to connect my [...]

Seems I have a similar issue. I'm trying to expose the data port to socat, so it will become a virtual com port. This currently immediately 'breaks' communication to port 8888 of the ESP01 and won't respond any longer. At that moment the TFT35 can't control the SKR anymore either. I have to reset the SKR1.3 to get it working again.

Were you able to get a more stable connection?

p1ngb4ck, 2019-10-29:
@hapklaar unfortunately not - but I actually didnt have the time to go on any further on that topic .. will update this post with any further findings, as soon as I had the time to do more testing

sl1pkn07, 2020-04-25:
Done! (finally, i have the chips sinde october xd)

thanks @p1ngb4ck !

rlvanhuis, 2020-10-09:
No longer need this

closed 2020-10-09 by rlvanhuis