bigtreetech/BIGTREETECH-TFT35-V3.0 #4: 12864 Mode not responding

rlvanhuis, 2019-09-04:
What are the recommended Marlin 2.0 settings for the SKR Pro?
TMC 5160 SPI drivers installed.
I have tried several different 12864 code segments.  The emulated 12864 screen does not populate with any information.
Performed both hot boot and cold boot.  Same result.

jeremiahhale, 2019-09-05:
I'm getting similar results as @rlvanhuis is getting in testing the new FTF35 3.0... I've found that 12864 mode works with SKR 1.3 but not with the SKR PRO 1.1 board. Making matters worse TFT Touch does not work on either board. I've reflashed the latest firmware BIGTREE_TFT35_V3.0.21.bin and I'm running the BigTreeTech Version of Marlin on the SKR Pro 1.1 Board.

bigtreetech, 2019-09-10:
LCD12864 mode need set `#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER` in Marlin

extesy, 2019-09-10:
@bigtreetech what mode in Marlin should be set if TFT35 is used as a touch screen?

Mistergrow, 2019-09-10:
In Touch Mode you connect with serial so activate a serial Port in Marlin FW.

bakaufman, 2020-03-02:
rlvanhuis, ever get this figure this out? same problem on GTR

chronowerxs, 2020-03-09:
Has anybody found the solution for that issue?

bakaufman, 2020-03-10:
For me it was a bad screen. Second screen sorted it out. Returned first to amazon.

rlvanhuis, 2020-03-10:
No, got frustrated with the lack of response.  Shelved it hoping someone
might have working program after a year or so.  The latest v25 has comm
issues.  The printer stutters (SKR PRO V1.1) when connected over usb.
Right now just working over usb.
Thanks for asking.

ebraiman, 2020-08-03:
For SKR Pro V1.1 I created a video that should apply, https://youtu.be/9i3Ee6UtC-Q .

cengizceka, 2020-08-07:
skr 1.4 turbo board I tried everything, the result is the same problem continues, is there anybody who can solve this problem ???

ebraiman, 2020-08-10:
@cengizceka This might help ... https://youtu.be/v0PCzHGXTgk

oldman4U, 2020-10-09:
If someone has still issues connecting the TFT. Please go to the TFT firmware repository, read the ReadMe there and post a ticket if there is still a problem.

Thank you

Owner of this ticket 
Please help the community and close the ticket once you do not need it anymore. Thank you

rlvanhuis, 2020-10-09:
No longer need this ticket

closed 2020-10-09 by rlvanhuis