bigtreetech/BIGTREETECH-TFT35-V3.0 #6: Can the knob be used along with the touch screen?

gitxmax, 2019-09-08:
Please, make the both touch screen and the knob usable. Sometimes it's a lot easier to use the know, to make adjustments. Sad to see it's totally useless in serial mode.

spilz87, 2019-09-11:
I thought a sulotion like this too, 
But the problem is that knob is still connected to printer so when you will use it to make adjustments on touch screen, it will change the feed rate in background on your marlin ...

p1ngb4ck, 2019-09-11:
there is a big reason, why the functionalities are completely split. afaik a feature like this won`t be possible - at least without breaking the full dual-mode ability. it could theoretically be possible if you completely abandon 12864 mode, resolder the rotary encoder and change the firmware - does not really make sense I think.
Also : The tft mode doesn`t make sense for printer-calibration and setup from my pov. The 12864-mode is by far better for this usecase. Once u have set up everything well enough to print repeatedly by just starting the print without greater modifications - so that using the tft really makes sense - you don`t need the knob anymore. Beforehand - just use the 12864 mode for setup, it`s much more convenient this way anyhow - just my 5 cents

Sthopeless, 2019-09-11:
Wouldn't be possible to recreate marlins display and use it thought Serial connection instead of how is being done now?

spilz87, 2019-09-11:
What do you mean exactly ?
The goal of the lcd12864 mode is to be transparent to marlin and be able to act directly with low level of the printer. 
Using serial communication is not what I call low level : you can have a buffer overflow in marlin and so don’t be able to interact with the printer, but you still able to use knob and navigate in lcd12864 menus. 
A good exemple is : 
When you run a leveling, you can’t change the bed temperature through serial communication before the end of leveling, but you can do it through knob and lcd12864 menu.

Sthopeless, 2019-09-11:
@spilz87 Alright, that makes sense. I thought it would be better in the long run to have Marlin's display cloned into the STM32 and use it has serial connection only.

spilz87, 2019-09-11:
If the goal of your idea is to relieve Marlin during a long print, I don’t think it’s necessary : In general Marlin still send data to LCD12864 in open loop (not waiting for reply) without issue. 
And it implies to modify Marlin, which is not a good idea I think

spilz87, 2019-09-11:
If you have time, check how Creality has to modify Marlin in several parts to be able to use their touchscreen in CR10S PRO. They modify Marlin a lot to be reliable in serial mode, and so you can not update marlin anymore ...

Sthopeless, 2019-09-11:
Relieve Marlin and add the possibility of adding more options into the menu (background and letter color change for example) the possibility of using the USB host and maybe use Marlin's display with touchscreen option or with the rotary encoder.

Maybe Marlin didn't need to be modified much if we added this somehow as a 'real' display option

Sthopeless, 2019-09-11:
Also like that, every other BIQU TFT could have 'Marlin's' display since it was being done via serial and with touchscreen buttons

spilz87, 2019-09-11:
I’m not sure to understand 
But did you notice that LCD12864 mode STM32 receive data in SPI slave mode and don’t send any data to marlin. 
STM32 doesn’t “understand” what it display, it just set pixel by pixel.

spilz87, 2019-09-11:
The goal of LCD12864 mode is to act like a display, not to use the serial port

extesy, 2019-09-11:
If the touch screen firmware is running as its own controller, why can't it listen to rotary encoder on one of its input pins and control it's own menus?

spilz87, 2019-09-11:
It could be if you don’t connect the connectors exp1 and exp2 but it become a basic serial touchscreen. TFT35 V3.0 has the specificity to have 2 modes. 
Try to add a rotary encoder on V2.0 will be easier and less confusing in use I think

OleUrgast, 2019-10-11:
I could understand the problem with Knob+Touch of there we're a simple Switch to change between 12864 - mode and touchmode while printing; so booth modes beiing switchable while printing. I would realy like that.
But as far I read, you have to use different firmware anyway to change the modes. Wich meens it dies not make sense to Connect serial and exp1/exp2 the same time. If you flashed Touchscreen, you only Connect serial. So the Knob cant interfeare with Marlin. If you flashed 12864, you connect exp1/exp2, but the Firmware  do not look at the Knob anyway (Marlin does).
Why should Marlin got irritated by the STM32 Sendung the knob also?
Or is It simply the hardwaredesign, the rotary encorder using the power from exp1/exp2 and if you connect the Display to a Ramps with 5V, you would need extra level shifter to avoid damaging the STM32?

So definitly not a firmware Problem, but a Missing hardware option. So next hardware should simply use some jumpers to connect encoder to STM32 or exp1/Exp2, depending of the mode you flash.

spilz87, 2019-10-11:
@OleUrgast im not sur to understand what you say.
What board are you talking about ? V2 or V3 ?

In V3 : 
- It’s not 2 different firmwares for touch mode and 12864lcd mode, but 2 parts of the same firmware, you switch by a long press on knob.
- you don’t have to flash new firmware to switch 
- so it make sense to connect serial and exp1/exp2 when you switch.
-> so knob is always physically connected to mainboard and marlin and a long press and rotary to select which mode acts on marlin in background which is not good.

I change that on my code like this : 
- I replace menu action «settings / pc disconnect » to switch to 12864lcd mode
- in 12864lcd mode, any touch on screen switch to touchscreen 

I don’t know for V2

ice2009, 2019-12-22:
Hello :) 

**[PAID JOB]**I find developer who reedit firmware to add knob function to touch screen mode (You see on prusa mini ) Color display + knob. More information i told you via mail. When you interested please contacted me on mail mraz.mario28[at]gmail.com 

Thank's

Sthopeless, 2019-12-22:
@ice2009 will he also add Marlin's mode to all TFT versions?

ice2009, 2019-12-22:
@Sthopeless for tft35 v3 it is posibble reedit firmware ? Please send me mail :)

spilz87, 2019-12-22:
Not sure to understand what function did you had, can you explain it

ice2009, 2019-12-22:
i want use full color display (touch screen mode) but i want dissable touch function and control display via knob. You see function on prusa mini (Full color display + knob)

spilz87, 2019-12-22:
Ok but in this case you connect only serial port, true?

ice2009, 2019-12-22:
@spilz87 what do you mean ? Please contact me on mail :)

spilz87, 2019-12-22:
I mean on prisa mini it’s a serial screen with knob, it’s not the knob used by marlin connected on exp1 and exp2

spilz87, 2019-12-22:
And if I remember well, on prusa mini, it’s only tft without any touch possibility

ice2009, 2019-12-22:
Yes and this display i want create. It possible reedit tft35 v3 display ? Or must be create new pcb with display to serial connect ? It is possible told with you via mail or skype ?

spilz87, 2019-12-22:
Yes you can do it with tft35 v3 display, you have “Only” to re-edit the software

ice2009, 2019-12-22:
Would you know reedit software when I pay you?

ice2009, 2019-12-22:
@spilz87 can you send me mail please? mraz.mario28[at]gmail.com

alagerqvist, 2020-02-13:
I have made changes to the code to allow for navigation using the knob/wheel/encoder.
I have only made it possible to do that in the way the menu's are currently designed.
When using the knob the UI can be designed in another better way.

ice2009, 2020-02-14:
@al

> I have made changes to the code to allow for navigation using the knob/wheel/encoder.
> I have only made it possible to do that in the way the menu's are currently designed.
> When using the knob the UI can be designed in another better way.

Hello :) it is possible send me mail ? :) mraz.mario28(@)gmail.com ?

closed 2020-10-09 by gitxmax