bigtreetech/BIGTREETECH-TFT35-V3.0 #7: TFT connection + EXT1/EXT2 both not working : SKR1.3

coreflake1, 2019-09-10:
I'm having issues connecting the screen (TFT35) to a SKR 1.3 motherboard with TMC2209 stepper drivers in UART mode.
I have set 
```
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
#define SERIAL_PORT 0
#define SERIAL_PORT_2 -1
```
Tried swapping the serial ports, but no luck. 
If i connect EXT1 & EXT2 i get nothing black screen not even blacklight. 
If i connect serial i get backlight and "LCD12864 simulator" but no image on the screen.

if i disable 
`#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`

and enable 
`#define CR10_STOCKDISPLAY`

then connect the screen to EXT1 (motherboard) and EXT3 (screen) the screen is working in LCD12864 mode. 

I'm using the latest version of Marlin + Latest firmware version for screen from here.
What am i doing wrong here ?

coreflake1, 2019-09-10:
Small update : 

Got this crazy idea that the EXT1 and EXT2 might be swapped on my TFT35, I was correct. 

Connect EXT1 (motheboard) to EXT2 (TFT35) and connect EXT2(motheboard) to EXT1(TFT35) then LCD12864 simulation works by setting : 
`#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`

But no touch screen when connecting serial, only LCD12864 simulation seems to work. 
Any ideas how to get the touch screen working ?

BobP2020, 2019-09-10:
I have a setup similar to yours except I am using TMC2208 in UART mode. I think 2208 VS 2209 should make no difference in the operation. As far as connections and Marlin configuration I have SKR1.3 EXP-1 connected to TFT35 EXP-1 and SKR1.3 EXP-2 connected to TFT35 EXP-2 using the cables provided by BTT with no modification to the cables. Using the supplied cable (black one) I have the TFT35 "RS232" connector connected to the TFT pins on the SKR 1.3 board using the cable supplied by BTT. When installing this cable I was provided onto the SKR 1.3, the "printing" on the cable is facing the closest edge of the board. The lead that words printed on is on the left-most pin of the "TFT" connector (closest to the micro-SD socket). As far as the Marlin configuration I have made the definitions similar to what you first published except my Serial  and Serial_2 are reversed (I do not believe this makes any difference).  So the definitions are like this-
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
#define SERIAL_PORT -1
#define SERIAL_PORT_2 0
Both the 12864 emulation mode and the touch screen mode work as expected for me with this rather  "standard" setup.

coreflake1, 2019-09-11:
Stupid me got it working. 
Didn't realise that you had to initiate the touch screen mode by pressing down the knob 10 sec.
Thanks for the help :)

closed 2019-09-11 by coreflake1