bigtreetech/BIGTREETECH-TFT35-V3.0 #9: U-Drive kills TFT35 when used with FW version .23

BobP2020, 2019-09-14:
When a flash drive is inserted into the USB socket of the TFT35 V3.0 running the .23 firmware the TFT35 stops responding to all commands/input. Version .21 of the firmware did not experience this problem.

brasco901, 2019-10-03:
This is happening to me as well.  Is there a fix for this yet?

BobP2020, 2019-10-03:
Try updating your TFT firmware to the very, very latest 0.23 version. It must be dated 20 September 2019 or later. The 0.23 version source that was recently released had problems as I described and I believe issues similar to what you are experiencing. I was directed to this newer release of the 0.23 firmware by the BTT staff. Important that the date is 20 Sept. or later as they did not change the rev., it is still 0.23 (makes it a bit confusing).

BobP2020, 2019-10-03:
This should be the link to the corrected 0.23 firmware----https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/blob/master/BIGTREETECH%20TFT35%20V3.0/Firmware/BIGTREE_TFT35_V3.0.23.bin

closed 2019-10-03 by BobP2020

brasco901, 2019-10-03:
That's the firmware I'm using and it still doesn't work

BobP2020, 2019-10-03:
Hmmm. That bin file fixed most all issues for me. When you bring up the "info" screen on the TFT display does it report a date of September 20, 2019?

brasco901, 2019-10-03:
No...when I updated...it has a date of sep 18th.  ???  gonna try updating one more time

brasco901, 2019-10-03:
This is the date I get... Even from the direct link you posted
[image]

brasco901, 2019-10-03:
also...when printing from the touchscreen...the led's don't work or change color according to progress.  with the lcd12864 it does.  any ideas about that?

BobP2020, 2019-10-03:
OK...not good. Let me see if I can re discover the link provided by BTT-
[image]

This is my attempt to include the actual image sent to me by BTT. First time trying this so it may not work.

BobP2020, 2019-10-03:
Above is the response I got from BTT via messenger. Some how I was able to navigate to this location and that .bin file gave a Sept 20 date.

BobP2020, 2019-10-03:
Try this link-   https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update

I don't know why it would be different other than poor housekeeping by BTT

brasco901, 2019-10-03:
I will try that when I get back home in an hour

BobP2020, 2019-10-03:
Yes, this link look exactly like the picture sent to me by Zyfio of BTT so I think this one should work (hopefully).

brasco901, 2019-10-03:
Ok.   Can you ask them via messenger any my led strip doesn't respond in touchscreen side?

brasco901, 2019-10-03:
I have the sept 20 date now.  I think they need to add that to this section to be consistent https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/tree/master/BIGTREETECH%20TFT35%20V3.0/Firmware

brasco901, 2019-10-03:
I have my skr v1.3 set up to use neopixel led lights.   It gives me indication when the heat bed and hotend are heating by changing colors.  It also provides white light to iluminate the print bed for me when it's printing...then the lights turn green when the print is finished.  Everything works fine when using the TFT35 V3 screen in lcd12864 mode.  The led lights do not work in serial/touchscreen mode.  Any idea why it doesn't?

brasco901, 2019-10-04:
?

BobP2020, 2019-10-05:
I have never worked with the LED as you describe. Sorry cannot be of help here. I would suggest presenting this issue on BigTreeTech's Facebook group. That is where I have received a lot of useful information and there are specific BTT staff assigned to monitor and respond to comments posted there. In particular, BTT Zyfio is the person that proved the TFT3 V3.0 information.

brasco901, 2019-10-05:
i requested to join the facebook group yesterday but still haven't been approved....patiently waiting so i can pose my question there.  thank you for everything BobP2020

brasco901, 2019-10-07:
Nobody is responding to my request to be added to the fb group 😒😕

BobP2020, 2019-10-07:
hmmm. I seem to remember a mention of a national holiday in beginning of October. Perhaps that would explain the delay.

BobP2020, 2019-10-07:
This is a message from the Facebook group- "Hello,October 1st is China's National Day, and this year marks the 70th anniversary of the founding of the People's Republic of China.
During this period, we will have a holiday from October 1st to 5th. We will officially go to work on October 6th. Therefore, please forgive us for not replying to your information in time during the holiday.😊"      So they should begin playing catchup today.