bigtreetech/BIGTREETECH-TFT35-V3.0 #10: TFT35 V3.0 gcode file format not compatible with ESP3D 8.3 restriction

BobP2020, 2019-09-15:
It appears that the TFT35 V3.0 gcode file format requires the file extension to be ".gcode". However, if one wishes to use the ESP3D utility with the WiFi module there appears to be a filename format incompatibility as the ESP3D requires an 8.3 format. It would be nice if the TFT35 V3.0 firmware would accept an extension of ".gco" as well as ".gcode". As it stands now I cannot use the ESP3D UI to start a print job.
In addition, (and it might be related?) I cannot "Upload" a gcode file using the ESP3D UI. The file upload appears to start but after about 15% it stops with and "Error 0" message. The TFT35 V3.0 display then shows a message in red print at the top of the screen that reads "Busy processing, please wait..." .  The TFT35 display is now non-operational. The touch functions allow changing screens but no commands are sent to the printer.

hapklaar, 2019-10-25:
Having a similar issue where commands sent to TFT35 v3.0 via ESP01 (wifi) module eventually don't reach the SKR1.3 anymore. Seems like a software issue on TFT35 to me.

At that point move commands on TFT35 do nothing anymore and also get the "Busy processing, please wait..." line. Only a reset of the TFT35 restores functionality.

oldman4U, 2020-06-12:
Hello BobP2020.

Have you tried the latest versions of firmware? Using gcode files with ESP should work now. Please check and let me know

Thank you

oldman4U, 2020-06-21:
@bigtreetech Could you please close this almost 1 year old ticket

This could help new users find tickets which already cover their topic.

Thank you

closed 2020-06-22 by BobP2020

Ohmarinus, 2020-08-03:
I am using the latest version of the TFT 35 firmware and am still running into this issue sadly. Any fix?