bigtreetech/BIGTREETECH-TFT35-V3.0 #13: TFT35 V3 custom menu items

brasco901, 2019-09-24:
Is there a way to add additional pages/menu items to the touchscreen?

brasco901, 2019-09-24:
Anyone?

coreflake1, 2019-09-29:
[](https://github.com/bigtreetech/BIGTREETECH-TFT35-V1.2/blob/master/TFT%2035%20screen%20customize.pdf)

Please let me know if you make something :). 
I think the graphics on the screen are horrible.

extesy, 2019-09-30:
@coreflake1 This guide only explains how to replace _existing_ icons, there is no information about adding new items or pages.

GenaUser, 2019-10-01:
I need this possibility too.
Currently, I am forced to use MKS screens because of this.

thisiskeithb, 2019-10-26:
> Please let me know if you make something :).

Check out these icons: [tft35-v3-0-iconpack](https://theveel.com/sdm_downloads/tft35-v3-0-iconpack/)

> This guide only explains how to replace existing icons, there is no information about adding new items or pages.

You'll have to take a look a the existing code. There's no drag & drop/easy way to add new buttons or screens.

closed 2020-10-23 by brasco901