bigtreetech/BIGTREETECH-TFT35-V3.0 #14: No printer attached on serial mode

emilhse, 2019-09-25:
When you switch between mode 12864 mode and serial mode  gives error. on 12864 mode it says printer is ready, on serial mode it says no Printer attached. updated the firmware same error is initiated

emilhse, 2019-09-27:
seems like i found something
it's a firmware mistake, it initiates error with out checking

BIGTREETECH-TouchScreenFirmware-master\TFT\src\User\Menu\menu.c
line 63: static REMINDER reminder = {{0, 0, LCD_WIDTH, TITLE_END_Y}, 0, STATUS_UNCONNECT, LABEL_UNCONNECTED}; there is no if or else statement so it will be on the screen forever.

BIGTREETECH-TouchScreenFirmware-master\TFT\src\User\API\Language\language_en.h
Line 72: #define EN_UNCONNECTED "No printer attached!"

Walstad83, 2019-10-11:
how does it work now?

emilhse, 2019-11-29:
Everything is ok now. I got it working

closed 2019-11-29 by emilhse

TheDoc97, 2020-03-07:
I think it is a difference in baudrate. Printer connected at 115200 while flashed firmware baudrate said 250000. After lower via display printer showed connected

obsidiansro, 2024-08-05:
Hi,

I am having an issue with

No printer attached!

Printer: Creality Ender 3 V2 with BL Touch
Board: SKR Mini E3 V3.0
Display: TFT35 V3.0

It was working with SKR Mini E3 V2.0 but after upgrade to SKR Mini E3 V3.0 board and display refuse to communicate.
I have tried to change Serial ports and matching baud speed with no success.

I would appreciate if someone with same setup could provide correct config advice.