bigtreetech/BIGTREETECH-TFT35-V3.0 #15: BIGTREETECH TFT35 V3.0 mode 12854 background color

vMeph, 2019-09-25:
was wondering about BIGTREETECH TFT35 V3.0 in mode 12854 all pictures i see the background is black and text green.
is there any options then to change background colors?

thisiskeithb, 2019-10-26:
Update your TFT firmware to the latest version. It allows you to change the colors in the config or through the TFT interface/on the fly without recompiling now.

vMeph, 2019-10-26:
@thisiskeithb  that seems to work for the touch mode, but not working for the 12854 mode

```
#define ST7920_BKCOLOR BLACK
#define ST7920_FNCOLOR BLUE
```
works tiuch mode only
the 12854 mode is always black and green

thisiskeithb, 2019-10-26:
Use the firmware from the [TouchScreenFirmware repo](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware) and not this one. I don't know why BigTreeTech has multiple copies.

vMeph, 2019-10-26:
i did use that repository......

vMeph, 2019-10-26:
i have opened a issue on that repository to

thisiskeithb, 2019-10-26:
In TFT mode, does this work?

> setting -> more set -> Change 12864 background color
> setting -> more set -> Change 12864 font color

Then switch to 12864 mode and test.

vMeph, 2019-10-26:
doing by the use of the touch you change colors on buttons but wont do nothing
[image]
but if you change in config file code then works on touch but on  12864 is always green
[image]

vMeph, 2019-10-26:
the firmware im using is the one you mention
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

thisiskeithb, 2019-10-26:
Mine doesn't do that, so I don't know.

thisiskeithb, 2019-10-26:
What does Settings-Info report?

vMeph, 2019-10-26:
board: bigtreetch_tft35_v3.0
firmware: tft35_v3.0.24.1 oct 21 2019

thisiskeithb, 2019-10-26:
Strange indeed.

vMeph, 2019-10-26:
wow it semmed to have worked now
os so before i have edit the conf file ant what i changed was
`#define ST7920_BANNER_TEXT ""` that was to remove the text LCD12864 Simulator
and
```
#define ST7920_BKCOLOR BLACK
#define ST7920_FNCOLOR BLUE
```
so now i just recompile all from scratch and is working.......

vMeph, 2019-10-26:
now the question is how do i remove the text ``LCD12864 Simulator`` from the lcd qith out recompile the bin file to set
ST7920_BANNER_TEXT "`LCD12864 Simulato"
to
ST7920_BANNER_TEXT ""

thisiskeithb, 2019-10-26:
> now the question is how do i remove the text LCD12864 Simulator from the lcd with out recompile the bin

You'll have to recompile and set it to `ST7920_BANNER_TEXT ""`. I plan to put in another PR to make that more elegant.

Can you try the attached firmware on your TFT35 to see if the background and font colors change under 12864 mode?
[BIGTREE_TFT35_V3.0.24.1.bin.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/3774775/BIGTREE_TFT35_V3.0.24.1.bin.zip)

vMeph, 2019-10-26:
yes it worked but resolution is very huge :)

vMeph, 2019-10-26:
looks more nice with 
`//#define ST7920_FULLSCREEN`
for some reson at fullscreen does look good 
[image]

if you notice is to stretch the square of fan is not even so looks better with out the fullscreen

thisiskeithb, 2019-10-26:
> yes it worked but resolution is very huge :)

I like it full-screen mode 🙂 The aspect ratio is not correct in the code, so it's stretched vertically.

> so now i just recompile all from scratch and is working.......

I just saw this. Glad you got your copy working!

vMeph, 2019-10-26:
ya thanks alot

vMeph, 2019-10-26:
@thisiskeithb  ya would it be nice to make it bigger but with the correct aspect ratio
about the ST7920_BANNER_TEXT ""will be looking into see your PR, or why not have without text by default, makes no sense to have text anyway, at least from my point of view hehe
thanks alot for your help :)

closed 2019-10-26 by vMeph

pr0jektcha0s, 2020-02-19:
I'm having this same issue with 3.0.25.1. Trying to set colors to Black/Blue. I've enabled and disabled banner text and that made no difference. Any suggestions?

// Marlin Mode Background & Font Color Options
// Current color options from lcd.h: BLACK, BLUE, BROWN, BRRED, CYAN, GBLUE, GRAY, GREEN, MAGENTA, RED, WHITE, YELLOW
#define ST7920_BKCOLOR BLACK
#define ST7920_FNCOLOR BLUE

// Text displayed at the top of the LCD in Marlin Mode.
#define ST7920_BANNER_TEXT ""

// Run Marlin Mode fullscreen. Not recommended for TFT24.
#define ST7920_FULLSCREEN