bigtreetech/BIGTREETECH-TFT35-V3.0 #16: heat for bed and hotend not being read on serial when printing

brasco901, 2019-10-03:
Started a print via usb.  The bed gets hot, but the temp never changes on the display on the touchscreen. also the led lights do not respond to temp changes on startup. Anybody else experience this?

brasco901, 2019-10-03:
I reflashed .23 firmware.....everything seems to be working but the leds don't change color.  when i use the non-touch screen....the led lights behave like they should.  any ideas?

BobP2020, 2019-10-03:
I am not sure what LED color change you are referring to. Where are the LED located?

brasco901, 2019-10-03:
I have my skr v1.3 set up to use neopixel led lights.   It gives me indication when the heat bed and hotend are heating by changing colors.  It also provides white light to iluminate the print bed for me when it's printing...then the lights turn green when the print is finished.  Everything works fine when using the TFT35 V3 screen in lcd12864 mode.  The led lights do not work in serial/touchscreen mode.  Any idea why it doesn't?

closed 2019-10-03 by brasco901