bigtreetech/BIGTREETECH-TFT35-V3.0 #17: no led light function with tft35 v3 on skr 1.3

brasco901, 2019-10-03:
I have my skr v1.3 set up to use neopixel led lights.   It gives me indication when the heat bed and hotend are heating by changing colors.  It also provides white light to iluminate the print bed for me when it's printing...then the lights turn green when the print is finished.  Everything works fine when using the TFT35 V3 screen in lcd12864 mode.  The led lights do not work in serial/touchscreen mode.  Any idea why it doesn't?

stefi01, 2019-11-10:
i just installed some on my skr 1.3, the leds work fine when i take 5v to the mosfets but not through the skr board, i used all 3 max end stop pins, enabled in config h and reasigned the pins to max rgb but whn i went to the led menu and selected, nothing works, the multimeter across the signal pins indictats 0.36v

bakaufman, 2020-02-23:
does anyone have the LED lights working?

stefi01, 2020-02-23:
> does anyone have the LED lights working?

nope, only managed 1 color and that was by using the servo pin, p2_0 i think so i upgraded to the skr v1.4 and that got them working, sorta

merinkumar, 2020-07-12:
Do we have any solutions yet? Am confused, as why emulator mode leds work and not in touch

stefi01, 2020-07-12:
same here, emulator works but touch just says no printer conected so i use emulator, if it works, dont break it, like they keep doing

avidNewb, 2020-09-23:
I have the same problem.  It seems to me that this is because the TFT just controls the board by sending serial commands which marlin doesn't use to run the neopixels. Either BTT needs to implement neopixel contorted into their firmware or marlin needs to implement neoplixel with serial commands.

closed 2020-10-23 by brasco901