bigtreetech/BIGTREETECH-TFT35-V3.0 #19: Size of screen info in 2.3 firmware on lcd12864 mode

brasco901, 2019-10-04:
Is there any way to increase the layout size on the .23 firmware like it is in the .21 firmware?  The information takes up the entire screen in .21,  but in .23 it's a smaller area of the screen when in lcd12864 mode

spilz87, 2019-10-11:
We have source files, it’s an option to enable or not. You have to compile it with your settings and options

brasco901, 2019-10-12:
Do you have instructions on how to enable?

spilz87, 2019-10-12:
Follow the instructions to compile for your board : 
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

The option you want is in BIGTREETECH-TouchScreenFirmware/TFT/src/User/Configuration.h
Uncomment : 
#define ST7920_FULLSCREEN

darap89, 2019-11-28:
CLOSE THIS ISSUE

oldman4U, 2020-06-12:
Hello brasco901.

Could you please close this ticket.

Thank you

oldman4U, 2020-06-15:
@bigtreetech Please close this unneeded ticket. Thank you

oldman4U, 2020-06-21:
@bigtreetech Could you please close this very old ticket unused ticket

This could help new users find tickets which already cover their topic.

Thank you

closed 2020-10-23 by brasco901