bigtreetech/BIGTREETECH-TFT35-V3.0 #20: TFT35 V3.0 Bootloader

0Jmaster0, 2019-10-14:
Hello and thank you for your schemes! I have problem - I accidentaly burned microcontroller of TFT35 v3.0 with my printer electronics. Now it shows no connection in stm32 st-link utility. The voltages of board (3v3 and 5v) seems to be normal and touchscreen seems to be undamaged, so I bought new STM32F207VET6 and soldered it instead of a dead one, now it is visible with ST-Link, but I found out that there is no bootloader file in repository. You gave bootloader for TFT V2.0, in one of the issues in "BIGTREETECH-TouchScreenFirmware" but for 3.0 it is not working(yes, I tested it :) I didn't seriosly think that it would work due to different chip)
Can I ask you to give bootloader file for V3.0? Thank you for answer

0Jmaster0, 2019-10-16:
Or, maybe, someone who have both TFT35 v3.0 and STLink can make full dump of microcontroller code and send it to me? All you need to do is to connect STLink to TFT, run STLink utility, click Target-Connect,  write size of microcontroller memory 256*1024 byte in HEX =0x40000 into field Size, press Enter, save dump File-Save file as and upload it into some of the file sharing sites like dropbox
Thank you!

Baltazart, 2019-10-16:
> Ou peut-être que quelqu'un qui possède à la fois TFT35 v3.0 et STLink peut effectuer une sauvegarde complète du code de microcontrôleur et l'envoyer à moi? Tout ce que vous avez à faire est de connecter STLink à TFT, d'exécuter l'utilitaire STLink, de cliquer sur Target-Connect, d'écrire la taille de la mémoire du microcontrôleur sur 256 * 1024 octets dans HEX = 0x40000 dans le champ Taille, appuyez sur Entrée, de sauvegarder le fichier enreg. il dans certains des sites de partage de fichiers comme dropbox
> Merci!

Hello, that's what you'll need, sorry for my Englishman. I speak only French.
If you guide me, this is the first time I use this software.

Baltazart, 2019-10-16:
> > TFT35 v3.0 et STLink peuvent effectuer une sauvegarde complète du code de microcontrôleur et l'envoyer à moi? Tout ce que vous avez à faire est de connecter STLink à TFT, d'exécuter l'utilitaire STLink, de cliquer sur Target-Connect, d'écrire la taille de la mémoire du microcontrôleur sur 256 * 1024 octets dans HEX = 0x40000 dans le Champ Taille, appuyez sur Entrée, sauvegardez le fichier enreg. il dans certains sites de partage de fichiers comme dropbox 
> > Merci!
> 
> Bonjour, c'est ce dont vous aurez besoin, désolé pour mon Anglais. Je parle seulement français.
> Si vous me guidez, c'est la première fois que j'utilise ce logiciel.

I prepare it for you

Baltazart, 2019-10-16:
> > > TFT35 v3.0 et STLink peuvent effectuer une sauvegarde complète du code de microcontrôleur et l'envoyer à moi? Tout ce que vous avez à faire est de connecter STLink à TFT, d'exécuter l'utilitaire STLink, de cliquer sur Target-Connect, d'écrire la taille de la mémoire du microcontrôleur sur 256 * 1024 octets dans HEX = 0x40000 dans le Champ Taille, enregistrée sur Entrée, sauvegardée le fichier enreg. il dans certains sites de partage de fichiers comme dropbox 
> > > Merci!
> > 
> > 
> > Bonjour, c'est ce dont vous aurez besoin, désolé pour mon anglais. Je parle seulement français. 
> > Si vous me guidez, c'est la première fois que j'utilise ce logiciel.
> 
> Je te le prépare

https://www.dropbox.com/sh/lrkgg90to23in0q/AABzS-Y83mZFh6yqejNspFeza?dl=0

Baltazart, 2019-10-16:
> Or, maybe, someone who have both TFT35 v3.0 and STLink can make full dump of microcontroller code and send it to me? All you need to do is to connect STLink to TFT, run STLink utility, click [...]

https://www.dropbox.com/sh/lrkgg90to23in0q/AABzS-Y83mZFh6yqejNspFeza?dl=0

0Jmaster0, 2019-10-16:
Thank you! It is working now! Simply flashed to my lcd through st-link and now everything is good again :)

closed 2019-10-16 by 0Jmaster0

thisiskeithb, 2019-11-02:
@0Jmaster0: Is there a trick to updating the firmware once you've flashed the bootloader with an ST-Link? I recovered a TFT35 3.0, but subsequent firmware updates using the SD card result in an ILLEGAL FLASH APP warning/unusable TFT.

0Jmaster0, 2019-11-02:
No. Only one problem was, that after dump flashing and display restarting, I received blue screen with no activity. But after I inserted sd card with firmware bin and TFT35 folder everything successfully updated and TFT is now working as before

Abyss8, 2022-03-20:
> Or, maybe, someone who have both TFT35 v3.0 and STLink can make full dump of microcontroller code and send it to me? All you need to do is to connect STLink to TFT, run STLink utility, click [...]

> https://www.dropbox.com/sh/ux8hzcrtyk7opqb/AABbvCSmki6Qc2MzMrquTzyna?dl=0

Can someone tell me how exacly connect STLink to TFT screen TFT35 3.0 or E3 ?

yigorello, 2023-10-14:
Hi all. Need a bootloader for TFT35-e3-v3.
Chip was replaced with a new one. Burnt out due to incorrect placement of EXP contacts

yigorello, 2023-10-17:
> Hi all. Need a bootloader for TFT35-e3-v3. Chip was replaced with a new one. Burnt out due to incorrect placement of EXP contacts (flipped 180 degrees)

I’ll answer myself. Found
This is bootloader tft35 GD.bin.
Everything is working :)