bigtreetech/BIGTREETECH-TFT35-V3.0 #21: LCD simulator bug after reset

shark-h44, 2019-10-26:
The display works well at first start, but bug after a reset or restart without power failure of several seconds.
**Screenshot of the display that works:**
[image]

**Capture lcd bug:**
[image]

**My config:**
- MKS SBASE V1.3
- Marlin 2.X bugfix
- Power supply 12V 300W

thisiskeithb, 2019-11-28:
Please try the latest firmware from the [main Touchscreen Firmware repo](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware) and see if your issue persists.

RomRider, 2019-12-05:
Running latest Firmware on the display (`Vx.x.25`)

I have a similar issue where nothing is displayed, then if I switch to touch mode and back to marlin mode, nothing is displayed until I click the button, and then it displays a bunch of random stuff like above, I can see pixels moving if I rotate the button or click on it (like if it tries to display some menu but fails). I never got it working. If I plug the original display with the exact same config it just works.

Board is SKR 1.3 running klipper, EXT1 on the board is connected to EXT3 on the display. Working in CR10 Mode.
Config is as follow in Klipper:
```
[display]
lcd_type: st7920
cs_pin: EXP1_7
sclk_pin: EXP1_6
sid_pin: EXP1_8
encoder_pins: ^EXP1_5, ^EXP1_3
click_pin: ^!EXP1_2
```

thisiskeithb, 2019-12-05:
> nothing is displayed until I click the button

That’s normal/expected behavior.

RomRider, 2019-12-05:
Thanks, I changed it, but it still displays random things.
[image]

poussmouss, 2019-12-20:
Same issue for me.
TFT35 V3.0 and SKR E3 DIP with TMC2209 both with latest firmwares.

No problem when the TFT is in BTT mode, only when in Marlin/LCD12864 simulator mode

At first it's like if the display was not refreshed fast enought, then these artifacts appear.

Sometime it's also a shifting of the display, about 10ps to the right and the right most pixels are displayed on the left.
I'll try to do some photos as soon as possible.

regards

pauloreal1, 2019-12-29:
I have the same problem with both my screens. I have the new version also that fits original casing, but this one has some vertical lines. So i installed my original lcd, this could be good but something is off

oldman4U, 2020-10-09:
Hi. What is the status of the reported issue?

pauloreal1, 2020-10-09:
Hello. It's all good now I have received a replacement

Enviado do Yahoo Mail para Android

oldman4U, 2020-10-09:
Cool. 

Whats about you @shark-h44!??