bigtreetech/BIGTREETECH-TFT35-V3.0 #23: Compiling .bin from Atom+PlatformIO not working

Ru83nG, 2019-11-12:
Followed all the steps for TFT35v3 and cannot get it to flash.
I set all parameters for my CR-10S on TFT>scr>user>Configuration.h
when I compile (Build) no errors.
I upload it to the \.pio\build\BIGTREE_TFT35_V3_0 folder.
Then I copy file BIGTREE_TFT35_V3.0.24.2.bin to root of SD card.
Then I add the TFT folder with all icons and fonts ....stick in the touch screen...power it up and nothing.
Then I keep the TFT file in the USB and add the current firmware from github...same bin file name.... and it works.
Any ideas? Need more info? I didn't want to add random information.
Thank you,

snadysand420, 2019-11-13:
I have the same problem with visual studio

BobP2020, 2019-11-13:
I had this same problem. After insert SD card into TFT35 V3 display it would show the usual % counting sequence but then all I had was a black screen, totally non-responsive. The problem was solved using a technique offered by the coding genius here on Github when working a similar problem with the SKR Mini E3 boards. The trick is to take "platformIO" back to an older version. I went back to core version 4.0.0 and this solved the problem for me. I am sure this is not the final solution but it works for now.

Ru83nG, 2019-11-14:
Thanks for your interest.
How is it you downgraded to v4.0.0?  
Not well versed in coding or this.  It's new to me.

I run this:  
pip install -U "platformio<4"
I get this:
PlatformIO Core v3.6.7 does not run under Python version 3.7.4 (tags/v3.7.4:e09359112e, Jul  8 2019, 20:34:20) [M
SC v.1916 64 bit (AMD64)].
    Minimum supported version is 2.7, please upgrade Python.
    Python 3 is not yet supported.
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
>> : The term '>>' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the
spelling of the name, or if a path was included, verify that the path is correct and try again.
At line:2 char:1

I installed python v2.7.17, restarted computer and still gives above error.
Do I need to uninstall Python 3.7.5 ?

Thank you,

Ru83nG, 2019-11-14:
Ok. I think I got it.
Unistalled Python version 3.7.4 
Ran:   pip install -U "platformio==4.0.0"
Restarted Atom, and now shows core 4.0.0
Thank you folks!
Now to try and build with my CR10S parameters.

Ru83nG, 2019-11-14:
Oh, and disabled automatic updates before restarting Atom

thisiskeithb, 2019-11-28:
PlatformIO no longer supports Atom, so any attempts to get it working require hacks and downgrades. Use VSCode or VSCodium instead as they're fully supported without downgrading.

The old firmware in this repo has been removed and users are now being pointed to the [main Touchscreen Firmware repo](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware).