bigtreetech/BIGTREETECH-TFT35-V3.0 #25: Display M117 om Statuscreen

uttrichp, 2019-12-07:
Hi There, 

is there a way to display the M117 responses from the Printer in the status screen in touch mode?

JWolfJr, 2020-01-04:
Would like to know also, I have my Ender 3 with a SKR mini E3, octoprint and now TFT35 V3. Can only get status on the simulated screen when printing from the Octopi.

tphan26363, 2020-04-25:
It displays a lot of OK pop up with irrelevant information, during printing such as  "Tool head 0! OK."  
I think we need some useful information to display but not asking to confirm every time.  Put all extra status update live in a separate status screen.

karabas2011, 2020-06-03:
I got M117 pop up during print, no problem. But print start confirm windows is UNDER "printing" buttons.
And yes, annoying Tool head X! Ok

leopshef, 2020-08-30:
Did this ever get added?

oldman4U, 2020-11-06:
M117 support has been added recently.