bigtreetech/BIGTREETECH-TFT35-V3.0 #27: In Unified Menu, Y+ and Y- both increase Y

jms07, 2020-01-05:
I have found that when using the Unified Menu, under Move, both Y+ and Y- cause the Y axis to increase by the requested amount. I have found the error to be in Move.c. The Case statements for Icon 1 (Y+) and Icon 5 (Y-) pass the same values to storeCmd.

I changed line 196 from "storeCmd(YGCODE_UP, item_move_len[item_move_len_i]);"
to "storeCmd(YGCODE_DOWN, item_move_len[item_move_len_i]);" and this resolved the issue for me.

I hope this helps for anyone else who runs into this issue.

[image]

oldman4U, 2020-06-15:
Hello jms07.

Could you please check with the latest firmware, your reported issue seems to be solved in the meantime.

Please close this ticket in case it is not needed anymore.

Thank you

jms07, 2020-06-22:
Yes, this issue has been resolved. Thank you oldman4U.

closed 2020-06-22 by jms07