bigtreetech/BIGTREETECH-TFT35-V3.0 #28: Revert back to original firmware

gl0zz3n, 2020-01-13:
Where would I be able to find the bin for the original firmware that comes on the screen when it is brand new? I would like to revert back as I lile that layout the best

Thi0114, 2020-01-27:
Have New ? I need that too :S thx

Thi0114, 2020-02-01:
Noone respond ? LOL... @bigtreetech

Thi0114, 2020-02-16:
Support from @bigtreetech  NICEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE

Thi0114, 2020-02-16:
@bigtreetech

gl0zz3n, 2020-02-16:
I actually used the master fw file from the github to compile my own,
custom version of the screen. It works really well for me now. Maybe you
will like it as well feel free to check it out https://youtu.be/5bkBtxbH7oA

oldman4U, 2020-06-12:
Hello gl0zz3n.

Seems you have been able to solve the reported issue in the meantime, so could you please close this ticket.

Thank you

oldman4U, 2020-06-21:
@bigtreetech Could you please close this unused ticket

This could help new users find tickets which already cover their topic.

Thank you