bigtreetech/BIGTREETECH-TFT35-V3.0 #29: unwanted extrusion

gadjet, 2020-01-27:
When selecting the extrusion menu in touch screen mode whilst printing and exiting out of it without pressing anything to extrude any filament my extruder extrudes a blob of filament all by itself!
I have an Ender 3 pro fitted with an SKR E3 mini 1.2 and a TFT35 V3.0.

The print is being sent from the connected Raspberry pi running Octoprint.

oldman4U, 2020-06-12:
Hello gadjet.

Could you please let us know the status of the reported issue. In case it has been solved in the meantime, please be so kind and close it.

Thank you

oldman4U, 2020-06-21:
@bigtreetech Could you please close this unused ticket

This could help new users find tickets which already cover their topic.

Thank you

gadjet, 2020-06-21:
To be honest @oldman4U I was waiting for someone at bigtreetech to answer my issue!
Is is a known issue/bug or not?
There's no point in being able to raise  issues with software if it's just ignored !!!

oldman4U, 2020-06-22:
Hi gadjet.

Sorry for the late answer. 3D printing is not my job...;-)

I use a SKR E3 DIP based Ender-3 with 2208's and a TFT35 E3 v3. So this is a slightly different hardware and i do not have this issue. Also that no other user answers your question makes me believe, that they do not have this issue too.

But i have a few questions.

When you click on Extrude while printing, do you get a warning that the print has to be paused because Extrude is not available while printing?
What would you like to do with Extrusion while printing (for me this function is very useless WHILE printing?
The time you have this unwanted extrusion, what is the temperature of the Nozzle?
Has the print already started?
Does the extruder motor turn exactly at that time?

Please let me know

Thank you

oldman4U, 2020-06-22:
Btw.

I am afraid, BTT will not answer your ticket because they "only" provide the hardware to this community and the community is responsible for the software. This is how Open Source works with all its pros and cons.

gadjet, 2020-06-22:
OK, I didn't realise the screen was open source, I though that was just Marlin.
My assumption was that it was software written by BTT.
Anyway it was that long ago I can't remember what I did or din't do so I'll just close it.

closed 2020-06-22 by gadjet