bigtreetech/BIGTREETECH-TFT35-V3.0 #30: Nis Gen L Btt tff35 e3 v3

Beaker1168, 2020-02-03:
I am needing guidance. I have hooked up via teaching techs video with it not working, when I updated to the latest firmware I then lost some of the icons was just a blank squares. Is there something that I need to do to Marlin? I am on an Ender 3 with Marlin 1.1.9

zelpha, 2020-02-05:
Sounds like you forgot to add the folder for the icons.

Beaker1168, 2020-02-05:
I have made sure I have added the folder to the Sd card each time, have tried both the unified and other both are given me issues.

Doug

Spidermakercnc, 2020-03-25:
Hi me if you want I can help I have your configuration and my tasks very well only the wifi I can't make it work

Clowntrigger, 2020-03-27:
See my reply at [https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/430#issuecomment-604664793].

You need to either rename the bmp files to the correct names or compile the latest firmware and install it.

oldman4U, 2020-06-12:
Hello Beaker1168.

Could you please close this ticket in case you do not need it anymore.

Thank you

Beaker1168, 2020-06-12:
Sorry not closing this earlier, not sure how but the 4th time was the charm and it all seems to be working now. thank you all of the help.

closed 2020-06-12 by Beaker1168