bigtreetech/BIGTREETECH-TFT35-V3.0 #31: BTT TFT35 V3.0 Ramps 1.4 White screen after sometime

wvabrinskas, 2020-02-05:
I have installed the latest version of the firmware and I can't for the life of me get the screen to stay on or remain functional. It seems that after sometime the screen will just turn white and remain that way even after a reset. 

#### Diagrams
I have the screen plugged into the proper spot like this : [image] and the EXT cables plugged in like this: [image]

#### Help
After attempting to start a print with an SD card it will start the process and then eventually just turn white and die. You can hear the buzzer go off every time to you tap the screen so there's something happening. Any suggestions?

wvabrinskas, 2020-02-05:
## Marlin config
`#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`
`#define SERIAL_PORT 0`
`#define BAUDRATE 250000`

Scope666, 2020-02-05:
One idea, try recompiling with baud at 115200 instead of 250000.  On mine I had bugs at 250000, no idea why.  ( I have increased buffers in my config )

wvabrinskas, 2020-02-05:
> One idea, try recompiling with baud at 115200 instead of 250000. On mine I had bugs at 250000, no idea why. ( I have increased buffers in my config )
> 
> It's in a few places, top of configuration.h and also in platformio.ini

I compile with the arduino IDE, what buffer setting are you changing in the configuration.h?

Scope666, 2020-02-05:
This is what I've been running for awhile now, they're actually the defaults now for my board.  (SKR Mini E3 v1.2)

[image]

wvabrinskas, 2020-02-05:
> This is what I've been running for awhile now, they're actually the defaults now for my board. (SKR Mini E3 v1.2)
> 
> [image]

Oh excellent thank you! Will give this a try. Just a heads up this in the `configuration_adv.h` file

Scope666, 2020-02-05:
Yes, sorry I forgot to mention where it was.  Remember the switch to 115200 is the important bit.  There's some weird interaction between the TFT and the SKR that causes issues for me at 250000.  It's worth a shot.

wvabrinskas, 2020-02-05:
So i tried this and the buffer stuff, 115200 seems to be the right baud rate, however the second i plug the SD in the screen goes white. Any ideas? it seems that the serial is busted?

Scope666, 2020-02-05:
What if you place a gcode file on a USB stick, and try to print it from the TFT, does that work?

Also maybe updating to the latest version of the TFT firmware, just to see if that changes anything.  There's been a lot of development / bug fixes recently.

wvabrinskas, 2020-02-05:
yeah ive updated as of yesterday. i'll double check and ill try USB

wvabrinskas, 2020-02-06:
@Scope666 no go. Updated, even got a new screen. It seems to be happening even more frequently now.

wvabrinskas, 2020-02-06:
So this is very odd. Plugging in the arduino mega to a raspberry pi running octo print and using an SD card on the BTT TFT35 worked....... the pi isn’t even connected to the printer. Wonder if it’s a power thing

wvabrinskas, 2020-02-09:
For anyone with this issue when using RAMPS 1.4 and an Arduino Mega, it is absolutely a power issue. The 5V rail on the supplied through the TFT port on the RAMPS is not enough and will drop out after a while. I solved this by plugging in the Arduino Mega through the USB port into a 5V wall transformer. Haven't had issue yet. 

Another thing to note: 
If you plan on controlling the printer via octoprint and plug in the serial cables via the wifi UART port on the BTT TFT35 If you plug it in to the Octopi and try to connect via UART you will run into a MIN_TEMP issue with one of the sensors as Octopi is reporting an inaccurate temp on the same bus as the TFT.

closed 2020-02-09 by wvabrinskas