bigtreetech/BIGTREETECH-TFT35-V3.0 #32: Reply V1.2 Automatic Shutdown Module connect to TFT35 V3 ?

joe-ko, 2020-02-11:
hallo,
i own an skr 1.4, the tft35 v3.0 and the the automagic shutdown module 1.2.
in the manual i readed that i can connect it but the Problem is that dont know how.
is it right that i can connect it to the tft ? the jst connector on the module has 2 Pins, the tft has 3.

oldman4U, 2020-06-12:
Hello joe-ko.

Have you been able to connect the Relay in the meantime?

Please let me know

Thank you

joe-ko, 2020-06-13:
No, send IT Back and got my Money back

oldman4U, 2020-06-13:
Thank you for the information. Could you please close this ticket then. Thank you

oldman4U, 2020-06-21:
@joe-ko 

Could you please close this ticket

This could help new users find tickets which already cover their topic.

Thank you