bigtreetech/BIGTREETECH-TFT35-V3.0 #35: Problem with BIGTREETECH TFT35 V3.0

Yotuntomatov, 2020-02-22:
Hello, I have a problem with the BIGTREETECH TFT35 V3.0.
BIGTREETECH TFT35 V3.0 in 12864LCD mode works without problems. In Touch screen mode, it constantly freezes, “unknown command M10 =”, “unknown command M105” and other errors that occur on any action from the screen are written on the screen. the seal hangs almost immediately and these errors pop up. Firmware for the latest version did not produce an effect.
Did you have the same problems, and how did you solve them? Thank you and sorry for my English.

Yotuntomatov, 2020-02-22:
TFT 35 ussed with SKR 1.3

nocturnalit, 2020-04-02:
I'm having the same problem, running SKR mini e3 1.2 and tft35 v3.0 (with firmware V3.0.25.3) the problem did not occur before updating firmware. I think it was previosuly running a November release.

fenix8k, 2020-04-09:
Same problem here skr 1.4 turbo, yesterday it was working fine, today I print a K calibration code, after 2 of them this start to happen

compositecrafters, 2020-06-06:
I have the same problem with the mini E3 board and TFT35 E3 V3 combo
screen stuck on Marlin or BigTree color mode! I mean stuck, stuck after updating to TFT35 E3 V3.0.26 stuck stuck, non respondooo!  tried several flashes still stuck
but the factory board works perfect, its like the TFT35 E3 V3 has a major Bug!

oldman4U, 2020-06-26:
Hi.

Has someone of you seen the reports regarding the metal frame touching the PCB of the TFT which can cause any kind of issue. Please check.

Thank you

oldman4U, 2020-08-18:
@Yotuntomatov 

Could you please let us know the status of the reported issue.

oldman4U, 2020-10-02:
Please help the community and close the ticket once you do not need it anymore. Thank you

Yotuntomatov, 2020-10-05:
I apologize for not answering, I didn’t know to close the question. The problem has been resolved. The error occurred due to a poor quality loop. After replacing the loop from another screen, everything works.

closed 2020-10-05 by Yotuntomatov