bigtreetech/BIGTREETECH-TFT35-V3.0 #36: Issue with printing from SKR mini E3 V1.2 on board SD card

Dimitar-88, 2020-02-28:
Hello guy's
I have and odd experience. The TFT35 V2 i have sees the on-board SD card of my SKR Mini E3 V1.2.
The issue is after i select a file to print i am taken to the screen to monitor the print but everything on the display side stops working let me put some colour to what happens:  
1.Display timer Starts counting. (that works)
2. Bed and Nozzle temps show 26/0 28/0 ( basically close to the room temp) The requested values do not get there and the actual never updates
3. Printer starts to heat up performs G28 and G29 accordingly and starts to print 

If i try and cancel the print the screen freezes and displays " Loading" and remains there.

I have  #define SDCARD_CONNECTION ONBOARD  (set)
and     #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER 

Before attempt to print i can manipulate all settings and the display works fine. have not tested to print from USB  or from LCD SD slot.

Everything works fine if i use the simulated graphical display.

jafc76, 2020-03-16:
I have exactly the same problem. It only happens if I'm on touchscreen mode. If I use the 12864 mode it works fine.

Bankinghelp, 2020-03-17:
@bigtreetech I have the exact same issue need fix now, very important!!!!!!!!!!

adzhsv, 2020-03-19:
I am also experiencing the exact same problems, Fan also stays at 0%

Bankinghelp, 2020-03-19:
@adzhsv @jafc76 @Dimitar-88  I have got the fix! Just change your #define SERIAL_PORT_2 to 0 in Configuration.h and #define SERIAL_PORT to -1!!! Problem solved ^^. Infinite loading isn't a problem, it's just that it has to reach max temperature until it cancels the print, so wait until then and it will stop loading! Hope this was useful!

Dimitar-88, 2020-03-20:
@Bankinghelp Setting up the SERIAL_PORT_2 to 0 is not possible !!! . The compiler throw's error.
In marlin firmware it is described that for MCU STM32F1 it is not allowed for Serial_port and Serial_port 2 to be of value 0

Dimitar-88, 2020-03-20:
@jafc76 @adzhsv @Bankinghelp  Just get latest firmware release 25.2 it fixed it for me

closed 2020-03-20 by Dimitar-88

adzhsv, 2020-03-22:
I still have this issue, I have the latest Firmware for the Both the SKR Mini and TFT35 v3.0

I have also tried changing the serial port, Then screen tells me No Printer Attached.

??

Dimitar-88, 2020-03-22:
@adzhsv I used Serial Port = - 1, Serual Port 2 = 2. And note in my first post both #defines are needed as far as i know.  When canceling the print you need to wait till last command finishes, but that's ok.

TheMiguelBi, 2020-04-30:
I'm also stuck here. 
Inserting a switch would cause the capacitors to remain charged.... 🤔