bigtreetech/BIGTREETECH-TFT35-V3.0 #40: SKR E3 1.2, TFT35-E3 v3.0 LCD Marlin Mode Scrabled Screen

wdjones23, 2020-03-04:
Looking for some help with skr 1.2 E3 TFT35-E3 v3.0 Ender5. Everything looks good on boot but when I heat it up and start moving the xyz axis the screen scrambles in marlin mode. Any ideas?

[image]
[image]

rc8181, 2020-04-19:
Having the same problem, did you find a solution?

oldman4U, 2020-05-13:
Hi.

Have you tried to use full screen mode? This works for me.

rc8181, 2020-05-13:
I did, and it didn't work for me. I ended up changing the board from the
mini e3 to the v1.4 turbo and it works fine.

wdjones23, 2020-05-14:
I found that it was a power supply issue. Changed to a Meanwell and
everything works as it should.

oldman4U, 2020-06-11:
Hello wdjones23

Could you please close this ticket. Seems the reported issue has been solved in the meantime

Thank you

oldman4U, 2020-06-21:
@bigtreetech Could you please close this ticket

This could help new users find tickets which already cover their topic.

Thank you

closed 2020-06-23 by wdjones23