bigtreetech/BIGTREETECH-TFT35-V3.0 #41: Screen locked on "chose mode"

Testmangh, 2020-03-08:
The printer was executing the command G29 (auto bed leveling) then the display went crazy it was on Marlim mode and it started  jumping to Big Tree mode then to marlin mode and so on, then it stopped on the screen where the user chose the mode, now even with a reset it is stuck on that screen, it doesn't even show the boot screen... It is stuck. Also it doesn't accept to rewrite the firmware using the SDCard, so I basically lost the LCD.

fenix8k, 2020-04-09:
Similar problem here, i can still use the marlin mode, try to disconnect the 5wire cable that enable the touch screen, going to try reflashing the firmware in a couple minutes

compositecrafters, 2020-06-06:
I just installed a new Mini E3 board and TFT35 E3 v3 it was working then it jumpped to the select Marlin or the Color menu and it stuck. no it is not shorting out on the metal frame, because its not installed in the metal frame. so the reset button is not being pushed, yes the reset button does resest it back to the stuck screen.
i have flashed it several times with no luck, once i got to a red dot configuration screen and it was stuck, so i did what they said and with a clean card placed a file with resest.txt and it did not work.
so I flashed it again and Im at the stuck Marlin or Bigtree color option and nothing works.
newest flash TFT35 v3.0.26

Jbchow, 2020-06-06:
I had this issue a few weeks back. For me the issue was the screws I used to mount it to my frame were too tight and the mount was pressing on the screen. Not sure if you installed it into a frame or not but if you did try loosening the screws and see if it goes back to working normally. If you press anywhere on the screen and hold there it goes in to choose mode as if you pressed the knob and held it.

cimoalpacino, 2020-07-07:
I had the exact same problem, when i tightened the screws too much.
If the bezel of your LCD cover is pressing on the touchscreen, your screen will always switch to the selection screen.
BR.

oldman4U, 2020-10-02:
Please help the community and close the ticket once you do not need it anymore. This topic is covered prominently in the ReadMe now. Thank you

oldman4U, 2020-10-05:
Please help the community and close the ticket once you do not need it anymore. Thank you

Testmangh, 2020-10-10:
The screws were too tight.

closed 2020-10-10 by Testmangh