bigtreetech/BIGTREETECH-TFT35-V3.0 #42: Extend the cable between the TFT and the board

Nicocostras, 2020-03-29:
Hello,
Is it possible to extend the cable between the TFT and the Board, approx 15 cm more ?
If yes, do i need to use a cable with a shield ?
Thank you

mpieroway, 2020-03-30:
are you talking the grey ribbon connectors for the EXP1 and EXP2 or the cable for the RS232?

RS232 I have extended using male to female arduino jumper cables.

For the ribbon if you get more ribbon cable you should be able to separate the connectors to replace the ribbon.

or purchase https://www.walmart.ca/en/ip/FC8P-100cm-IDC-8-Pin-Hard-Drive-Extension-Wire-Flat-Ribbon-Cable-for-Motherboard/PRD322QFC37V2RK

oldman4U, 2020-09-02:
Hello @Nicocostras.

Could you please close this ticket.

Thank you

closed 2020-09-03 by Nicocostras