bigtreetech/BIGTREETECH-TFT35-V3.0 #43: Some icons missing after firmware update

Stupendousman2017, 2020-03-30:
When I copy the firmware and TFT folder to the root, it starts to update normally.
After it boots, many of the icons are missing, and are replaced by a splotch of white dots, as if a snapshot of television snow.

Even when I built a firmware myself, it told me that a number of icons were simply not present in the directory.

What am I missing?

Clowntrigger, 2020-04-05:
Compile the firmware from the latest version available. Copy the .bin file to the root of the sd card along with the TFT35 directory for the version you're using (unified or standard). That worked for me and others are reporting success with this method.

SenilePOV, 2020-04-27:
Tore my hair out on this one, but I found the issue. In the version 26 firmware bundle for compiling a new bin file the API lists all the icons needed. The "missing" Icons are listed in the API as Status__Bed.bmp, Status_Fan.bmp, and Status _Nozzle.bmp. In the unified BMP folder these files are there listed at StatusBed.bmp, StatusFan.bmp, and StatusNozzle.bmp. Notice the difference. So to correct the issue I assume you can remove the underscore from the 3 names in the API section of the compiler and recompile, or you can do as I did and add the underscore to the files in the BMP folder.

jeffman78, 2020-05-09:
This fixed the issue. After recompiling the firmware using PlatformIO, I copied the new firmware on the SDCard together with a fresh copy of the TFT35 folder.
No error during import and after reboot, everything is now at it should be - all icons are there.
Thanks for your help.

oldman4U, 2020-06-12:
Hello Stupendousman2017.

Seems the reported issue has been solved, so could you please close this ticket.

Thank you

Stupendousman2017, 2020-06-12:
Thank you all
 Cheers!

closed 2020-06-12 by Stupendousman2017