bigtreetech/BIGTREETECH-TFT35-V3.0 #45: LCD12864 not showing after selecting

Tillema, 2020-04-06:
When I switch to the LCD12864 simulator, it only shows a blue screen with the text 'LCD12864 simulator'.
After rotating the rotary encoder, the menu of the simulator shows up, but it doesn't load the main screen without rotating the rotary encoder.

thisiskeithb, 2020-04-07:
[This is normal](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/blob/master/Hardware/Explanation%20of%20black%20screen%20in%20TFT35%20V3.0.pdf).

tony1ant, 2020-09-02:
Hi I'm having a similar problem,  but my TFT35 V3.0 only shows the text 'LCD12864 simulator' nothing else. If I turn the knob left or right nothing happens, it only shows the  text 'LCD12864 simulator'. I'm new to GitHub but I've been hobby 3D printing for  about 18 months now. I've tried different SD cards, re compiled the firmware numerous times & still the same result. I'm running Ender 3 Pro direct drive, with BTT SKR 1.4 Turbo & TMC2209 stepper drivers. Any help will be Greatly Appreciated. Thank U in advance.

oldman4U, 2020-09-02:
hello tony1ant.

This ticket is more about the fact, that you have to turn the encoder to get the screen active - after switching to Marlin mode. Your problem is a different one and i would like to ask you to start your own ticket.

Which kind of display have you defined in Configuration.h of your SKR v1.4? CR-10 or something else?

tony1ant, 2020-09-05:
Hiya oldman4U

Thank U 4 your reply, much appreciated. Turning the knob doesn't do anything. 
I'm new 2 GitHub and not sure how 2 start a ticket, but I will search the help section 2C how 2 do it. 

I hava SKR E3 V1.4Turbo—TMC22209's
BTT TFT35 E3 V3.0
Bl-Touch
Direct Drive (STD CE3 Hotend)
BTT UPS
BTT Power Off relay V1.2
DC to DC module
Filament Runout Sensor
NeoPixels 12 LED ring
Raspberry Pi 
then
REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER in configuration.h
&
CR10_STOCKDISPLAY in configuration.h

Thank You Very Much
Kind Regards
Tony1ant

oldman4U, 2020-09-05:
Hi. 

There is a NEW ISSUE button which allows to start a new issue. I do the 3D print stuff since a year now but i would not be able to configure all the hardware you have bought.

So the best way to start is to take the minimum hardware needed (Mainboard, drivers and TFT from the list above) and to add more once this works.

oldman4U, 2020-09-07:
@Tillema 

Please close this ticket. Tony started his own in the meantime.

Thank you

closed 2020-09-07 by Tillema