bigtreetech/BIGTREETECH-TFT35-V3.0 #49: echo: error unknown command ever 3 to 5 mins

carl5765, 2020-05-02:
im getting a echo error message every 3 to 5 min it beeps and message reads 
unknown command: "m73 p8 r220"
and 
m73 q8 s227
m73 p8 r221
m73 q1 6 s205 
help

oldman4U, 2020-05-13:
HI carl.

Would it be possible, that you use an extension in your slicer software, to show the layer high or progress?

This could eventually cause this issue.

oldman4U, 2020-06-12:
Hi,

Any news regarding the reported issue.

Could you please close this ticket in case you do not need it anymore

Thank you

Dingo1971, 2020-07-24:
I'm having the same issue, every so often ( 30 seconds to 2 minutes, varies) I get the same error message: see img below.
I know the M73 command is for print progress, 
I can't find anything in my slicer's setting that would make this happen.
I use PrusaSlic3r 2.2.0.
Any suggestions would be appreciated.  Thanks

[image]

oldman4U, 2020-07-24:
Check the gcode file itself.

 In Cura this is a script which has to be activated to see the layer number on lcd screen.

Dingo1971, 2020-07-25:
Thanks, I read through the gcode files and see an M73 line every so often,, 
But I cant find in Prusaslicer where to turn that on or off.
I'll have to do some more research on Prusa slicer.
Thanks for the help!!!

oldman4U, 2020-07-25:
👍🏻But carl seems to be a lazy guy. ;-)

@carl5765   Please close this ticket. Thank you

Dingo1971, 2020-07-26:
HI all, 
Just wanted to post an update, and I figured out where the extra Gcode was coming from in PrusaSlicer., for anyone else that may need to know.
It;s under PRINTER SETTINGS->GENERAL --FIRMWARE ---SUPPORTS REMAINING TIMES  un-check this option to kill the extra Gcode.
[image]

oldman4U, 2020-07-26:
Hi dingo. 

Great you found it. No idea if carl still watches his own ticket.

oldman4U, 2020-10-05:
Please help the community and close the ticket once you do not need it anymore. Thank you

Dingo1971, 2020-10-05:
> Please help the community and close the ticket once you do not need it anymore. Thank you

Sorry to be ignorant, but how do I close the ticket?

oldman4U, 2020-10-05:
Hi @Dingo1971. You are ok. Carl is ignorant and is the only one who can close this issue. Please delete your replies here to get no notifications any more. Thank you and sorry that I bothered you🥺

Dingo1971, 2020-10-05:
no worries,, is there away to keep my solution available so that others with the same issue can find it?

oldman4U, 2020-10-05:
It would be great if you could start a new ticket under the general TFT firmware repo describing the problem and the solution. This would make this information available not only for users of this particular TFT. THANK YOU 👍🏻

oldman4U, 2020-10-09:
@carl5765 

Please help the community and close the ticket once you do not need it anymore. Thank you

carl5765, 2020-10-10:
> 👍🏻But carl seems to be a lazy guy. ;-)
> 
> @carl5765 Please close this ticket. Thank you

lazy guys thanks great support answer

carl5765, 2020-10-10:
> 👍🏻But carl seems to be a lazy guy. ;-)
> 
> @carl5765 Please close this ticket. Thank you

ilazyzt your just plane rude

carl5765, 2020-10-10:
smart ass

closed 2020-10-10 by carl5765