bigtreetech/BIGTREETECH-TFT35-V3.0 #50: LCD Screen knob turns in wrong direction

Bamapaul, 2020-05-10:
Just installed and updated firmware. The knob in marlin mode turns the selector in the opposite direction as all my other printers. I cannot seem to figure out why. Any thoughts?

BobP2020, 2020-05-10:
This is configured within the Marlin firmware so you can change direction for menu select and increase/decrease direction

Bamapaul, 2020-05-10:
Is it in the printers firmware or the TFT screen  Sent from Mail for Windows 10 From: BobP2020Sent: Sunday, May 10, 2020 10:25 AMTo: bigtreetech/BIGTREETECH-TFT35-V3.0Cc: Bamapaul; AuthorSubject: Re: [bigtreetech/BIGTREETECH-TFT35-V3.0] LCD Screen knob turns in wrong direction (#50) This is configured within the Marlin firmware so you can change direction for menu select and increase/decrease direction

Bamapaul, 2020-05-10:
Found it! Thanks! That was waaayyyy too easy!

closed 2020-05-10 by Bamapaul