bigtreetech/BIGTREETECH-TFT35-V3.0 #52: Buzzer sound extremely low

BrotherBloat, 2020-05-11:
The buzzer sound seems extremely low - it's barely audible. Is there a setting that could override its volume? The sticker if off and the buzzer itself is exposed in my TFT case, but the sound is barely audible on any tone.

Is that normal on this screen?

steps to reproduce:
run the following melody GCode:
`M300 S2349 P53
M300 S0 P53
M300 S2349 P53
M300 S0 P53
M300 S2349 P53
M300 S0 P53
M300 S2349 P428
M300 S932 P428
M300 S2093 P428
M300 S2349 P107
M300 S0 P214
M300 S2093 P107
M300 S2349 P857`

Expected result: short tune audible over the sound of the fans running.
Actual result: barely audible, garbled audio coming from the buzzer on the back of the TFT.

AnHardt, 2020-05-11:
Try to use it as a BUZZER, not a SPEAKER.

BrotherBloat, 2020-05-12:
> Try to use it as a BUZZER, not a SPEAKER.

Thank you for the advice! From what I read this involves commenting out #define SPEAKER in the main SKR config, right?

giorgakos, 2020-05-26:
Have you tried removing the sticker of the buzzer?

BrotherBloat, 2020-05-27:
> Have you tried removing the sticker off the buzzer?

certainly - the sticker is off!

oldman4U, 2020-06-12:
Hello BrotherBloat.

Have you been able to solve the reported issue in the meantime and have you tried the latest version of TFT firmware which has a Buzzer related feature:

/**
 * The duration and frequency for the UI feedback sound.
 * Set these to 0 to disable audio feedback in the LCD menus.
 * Only valid for SERIAL_TSC LCD MODE and if BUZZER_PIN is set or available.
 *
 * Note: Test audio output with the G-Code:
 * M300 S<frequency Hz> P<duration ms>
 */
#define BUZZER_FREQUENCY_DURATION_MS 20 // Default 20
#define BUZZER_FREQUENCY_HZ 10000       // Default 10000, 20Hz to 60KHz
#define BUZZER_STOP_LEVEL false

Please let me know

oldman4U, 2020-07-24:
Hi.

I kindly ask you to close this ticket, as it seems that it is not needed anymore, thank you.

In the meantime I got 2 TFT35 E3 v3.0 and the buzzer sounds completely different on the two identical boards. So it could be easily possible that there are simply different buzzers used during production. Good thing is, that such buzzers are cheap and easily to source, so you could change it to get it louder.

oldman4U, 2020-08-18:
@bigtreetech 

Please close this ticket. Thank you