bigtreetech/BIGTREETECH-TFT35-V3.0 #54: ESP-01disables the ABL menu.

titus95, 2020-05-26:
When I install the ESP-01 wifi module, the ABL submenu no longer appears.

oldman4U, 2020-06-12:
Hello titus95. Have you been able to solve the reported issue in the meantime?

Please let me know

Thank you

titus95, 2020-06-12:
Hello, no, I didn't have time to take care of it. 
But there must be something wrong with Marlin.

oldman4U, 2020-06-22:
Are u using new versions of the firmware?

So you mean that connecting the Espies you can not see the ABL under Motion but disconnecting you can see it?

traffic-light, 2020-08-02:
I've got the same issue. 
When I reset the board I've got the menu back.

I think it's something with the serial port. The module communicates around the same time.

oldman4U, 2020-08-02:
I have only serial 2 defined.