bigtreetech/BIGTREETECH-TFT35-V3.0 #57: LED knob not working (in pre-compiled firmware)

dxduck, 2020-06-03:
Using the pre-compiled TFT_V3.0.26.x fw provided in the repo under (Copy to SD Card root directory to update - Unified Menu Material theme), the options to set LED knob color fail and result in echo message "M150 R255 U0 B0 P255" (result for red).

There is also no longer any option to turn on "Rotary Knob LED" in Menu-Settings-Features.

I had previously turned off the knob prior to upgrading from stock fw.

Additional question: is there a way to manually enable the knob LED through terminal command?

psampaiobh, 2020-06-09:
Having the same problem!

oldman4U, 2020-06-11:
Hello dxduck

I believe this is the wrong repository and your ticket should be opened in the general touchscreen repo or here: https://github.com/bigtreetech/BTT-TFT35-E3-V3.0

Anyway, the first issue should be fixed by using the Master Branch instead of firmware 26.x. Using Master you can define the LED Color or turn it off using the Master Branch itself and also using the TFT afterwards.

Do you need help to compile a Branch?

dxduck, 2020-06-12:
> Hello dxduck
>
> I believe this is the wrong repository and your ticket should be opened in the general touchscreen repo or here: https://github.com/bigtreetech/BTT-TFT35-E3-V3.0
>
> Anyway, the first [...]

Thanks oldman4u, I'll have to repost there. Was looking to avoid compiling altogether by using the pre-compiled bin. Wouldn't it be logical for that to be turned on in the pre-compiled bin specifically provided for the TFT35 E3 V3.0?

oldman4U, 2020-06-12:
Thats the disadvantage of open source. The frequency bugs are fixed or new
features are added depends on the number of users willing to spend their
valuable time to do whatever is needed. So the official binaries are always
weeks or even months behind the latest version. Compiling the TFT screen is
now even easier thanks to a text file where most functions can be changed
without compiling at all.

Next release should fix the issue you have reported.

oldman4U, 2020-06-12:
Sorry, mismatch in my head because i haven't used it for a while. Of course there is also a pre-compiled version available from here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware.git

Please install this version and let me know

Thank you

dxduck, 2020-06-12:
Thanks for your prompt replies, I appreciate your attention to the issue and your suggestions.

I was already using the latest firmware version from the link you provided (specfically, https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update ), which was also the most recent release (.26) and had flashed it twice just to make sure.

I ended up finding a working solution by using a config.ini file (Detailed instructions here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/config_instructions.md )

Setting the value for knob LED "knob_led_color:1" in config.ini worked. Once set to any value (except 0) in config.ini it becomes an active option in the Settings > Feature menu

IMO this should be enabled and set to a default of 1 both in the provided config.ini and also in the source code so that it's always a working feature. From what I've been able to gather it will be ignored on printers that don't support it so there shouldn't be any harm in setting it to a default activated value. I'll open a ticket in the proper forum.

oldman4U, 2020-06-12:
Yes. there are several versions with the same number, so it is always worth to check the Commits. 

Glad to hear i could help. The config.ini and the ability to dim the LCD after a defined time are my favourite functions;-)

Not sure why the default is off, but i am sure @guruathwal can answer this question for me.

Happy printing and please close this ticket once you do not need it anymore.

Thank you

closed 2020-06-12 by dxduck

xdased14x, 2020-11-10:
i know this ticket is closed just a quick question, tried the above fix in the config.ini and in firmware and neither worked is there something in marlin that i need to have set to allow these leds to work. thanks

oldman4U, 2020-11-10:
No reason not to use a ticket. Send a picture of the Info of the TFT.

jaroszmm, 2021-01-05:
I have the same issue - newest ready to use compilation but no way to enable LED. Even manually changed config.ini to =1 instead of 0 at the led knob color, but no effect.