bigtreetech/BIGTREETECH-TFT35-V3.0 #63: Print Does not end.

Dimitar-88, 2020-06-11:
Hey guy's. I have a problem that may have been discussed already but i fail to find a solution.
Here are the two scenarios followed by firmware version both happen in touch version and unified view .

1.  When i start a print from the onboard SD ( not the display SD or USB ) and then press stop ( before last version it was named cancel)
the display freezes with message " loading.... " and it stay's there for ever and the print continues until end of G code.

2. When i start printing from the onboard SD ( not the display SD or USB ) After the g-code end the display remains in printing screen % completion goes to 0 . The end g-code from the config file of the display is never executed and the auto shut down does not work as well. At that point if i hit cancel / stop print the screen gets again in " loading...." and never changes.

i am using SKR Mini E3 V1.2 and i have tried with latest release of the firmware
The display is running the latest version V3.0.26.x with all latest icons and others updated.

I had the same problem on older version as well. 

The thing i want most is the auto shutdown witch is connected to the LCD to work properly.

oldman4U, 2020-06-12:
Hello Dimitar-88.

There has been a PR 19h ago which should at least solve your temperature problem. Could you please check.

Regarding the other issues. I would recommend to start the same ticket but under the general BTT Touchscreen Repository and not here. This repository is not covered very well and i believe you will find a ticket related to the inability to stop a print there already.

Please let me know

oldman4U, 2020-06-21:
Hello.

Any news regarding the reported issue?

Please let us know

Thank you

oldman4U, 2020-06-26:
@Dimitar-88 

Could you please be so kind and answer or close this ticket.

Thank you

Dimitar-88, 2020-07-01:
@oldman4U
Yes temps seem to work . The other hang issues remain. i see even fan speed now displays ok so

closed 2020-07-01 by Dimitar-88