bigtreetech/BIGTREETECH-TFT35-V3.0 #64: Will not update firmware

sonny-c, 2020-06-12:
About a week and a half ago, I updated my SKR Mini E3 V1.2 & TFT35 E3 V3.0 to Marlin 2.0.5.3 by placing the compiled firmware.bin file in the root directory of my sd card.  I put the sd card into the TFT35 E3 V3.0 and powered up my Ender 3.  BOOM!  It updated with no issues.  Then I purchased (https://theveel.com/produkt/4-pack-sjo-rodgul-mork-lime/) and updated the icons repeating the same process.

Yesterday, I re-compiled Marlin 2.0.5.3 to add some additional features.  After placing the sd card into the TFT35 E3 V3.0, nothing happens.  So I tried another sd card, the same issue.  In total, I tried three different sd cards, 2GB, 8GB, and 32GB.  All the sd cards are SanDisk, and all were formatted to FAT32.  I even tried using the micro sd card that came with my Ender 3, but that did not work either.

All the sd cards work fine when using them to print code files.  Does anyone know why SKR or the TFT won't see and update the firmware.bin file?

oldman4U, 2020-06-12:
Hi sonny-c.

The firmware.bin file is for SKR boards only. Firmware for TFT's is always named like the TFT the firmware is made for, for example: BIGTREE_TFT35_V3.0_E3.26.x.bin

Make sure you add the corresponding TFT35 folder which holds the font and the bmp folder as well as the config.ini text file which you can use to change values without recompiling. All those files and folders will be renamed to CUR after the successful update.

Hope this helps, please do not forget to close tickets you do not need anymore.

Thank you

sonny-c, 2020-06-12:
> Hi sonny-c.
>
> The firmware.bin file is for SKR boards only. Firmware for TFT's is always named like the TFT the firmware is made for, for example: BIGTREE_TFT35_V3.0_E3.26.x.bin
>
> Make sure you add [...]

To make sure I understand, you are saying I cannot update the firmware of the SKR board using the TFT35 sd card slot?  I placed the firmware.bin file onto the micro sdcard and inserted the micro sd card into the printer's sd slot, not the TFT35.  Then I powered on the printer, but nothing happens.  Do the TFT35 disable the printer's micro sd slot?

Thank you for the updating procedure refresh.  My silly self only copied the bmp and font folders to the root directory.  What I needed to do is copy the TFT35 folder, that contains the bmp and font folders, to the root directory.

closed 2020-06-12 by sonny-c