bigtreetech/BIGTREETECH-TFT35-V3.0 #65: Mesh Bed Leveling not in Touch Mode

sonny-c, 2020-06-15:
I have the SKR Mini E3 v1.2 & TFT35 E3 V3.0 installed in my Ender 3.  Marlin 2.0.5.3 is the firmware and I enabled Mesh Bed Leveling.

I only see MBL in LCD mode and not in Touch Mode.  The only leveling options I see in Touch Mode are ABL and 4 Corner Leveling.

Is there a way to enable MBL in Touch Mode?

oldman4U, 2020-06-15:
Hello sonny-c.

I would recommend to close this ticket here and to open the same ticket in the generic BTT TouchScreen area. It is much more activity there.

sonny-c, 2020-06-15:
> Hello sonny-c.
> 
> I would recommend to close this ticket here and to open the same ticket in the generic BTT TouchScreen area. It is much more activity there.

Will do.  Thank you sir!  I'll post it in BIGTREETECH-TouchScreenFirmware.

closed 2020-06-15 by sonny-c

teknokraze, 2021-04-16:
Well, that just made this result in the Google search useless. Least you guys could've did was place a link here to the other ticket.

BoLtWD, 2021-11-14:
This is something I needed now cant find

oldman4U, 2021-11-14:
So you found this completely outdated ticket which is closed since a few months but you can nor find the BTT TFT Firmware page here on GitHub!?

BoLtWD, 2021-11-14:
Firmware page is not a problem just wanted to know where I could find the resolve of the issue above, thank you

oldman4U, 2021-11-14:
Mesh leveling in touchmode is part of the current firmware v.27.

PEZ18, 2023-05-12:
I cant find any help on this issue either. It works perfectly in Marlin mode but not in touch.  If anyone has a link to how to fix this please let me know.