bigtreetech/BIGTREETECH-TFT35-V3.0 #66: Display Fades after powered on.

N5L04, 2020-06-17:
I have a BTT TFT35-E3-v3.0 hooked up to a BTT SKR 1.4 Turbo. I followed the online manual to connect all the wires. When my printer is powered on the display will be bright then fades to black within a minute. If I power cycle the printer immediately the display stays dark. If I turn it back on after 20 minutes the display is bright then fades again.

oldman4U, 2020-06-20:
Hi.

Do you have the display installed in a metal housing? If so, make sure the display does not touch the frame. Best would be to try the same with the display unmounted and only the cable connected

Let me know

oldman4U, 2020-06-25:
Hi.

Any update on the reported issue?

N5L04, 2020-06-27:
sorry I figured out that the BLTouch i was using had a short in it. I replaced it and the display is working correctly now. Thank you.

closed 2020-06-27 by N5L04