bigtreetech/BIGTREETECH-TFT35-V3.0 #67: ESP-01S Istallation Issue

KL-XL, 2020-06-18:
Hi there,

I am a complete newbie in 3d printers. Just got the BTT TFT35 E3 V3.0, two ESP-01S modules and BTT Writer Module from BIGTREETECH Global Store at AliExpress. The TFT35 installed and operates smoothly, no issues. Having trouble installing ESP-01S. I followed all the steps in the manual:
- successfully flashed firmware using Writer Module, esp3d.bin file and flash_download_tools_v3.6.7 software
- downloaded the three files for upload: 404.htm, favicon.ico and index.html.gz
- placed ESP-01S on the TFT35 board
- got the IP 192.168.0.1
- connected my laptop to ESP3D network using the password 12345678
- opened 192.168.0.1 in MS Edge and got the initial file upload interface
- uploaded the three files and pressed refresh, the files appeared in the uploaded list (so far so good)
Here is where I got the issue: the Configuration Wizard didn't pop up. I waited for a while and the refreshed the page. The page turned blank. Now no matter what browser I use the 192.168.0.1 always shows blank. If I open it in IE then I can see the favicon in the address bar which sort of confirms that I see the right page and that the files have uploaded. However I am unable to progress nor revert to the initial file upload interface. The module still shows the IP when I turn on the printer and I can connect to ESP3D network no problem.
I am clueless as to if the modules are faulty or it a software issue. 
Any guidance is highly appreciated?

Thanks,
KL

My configuration:
machine: Ender 3
board: BTT SKR Mini E3 V2.0
display: BTT TFT35 E3 V3.0
OS: Windows 10
slicer: Cura 4.6.1
other: Pronterface/Printrun 1.6.0
[image]
[image]
[image]

oldman4U, 2020-06-20:
Hi.

I guess you uploaded a firmware which was too big for your Espies module. So i would recommend to reset the board like described here:
[image]

Once you are back to the page, where you can upload firmware, take the firmware from here as a starting point: https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/tree/master/firmware/esp-01s

Let me know

KL-XL, 2020-06-20:
Thank you oldman4U. I am able to see the upload page now. So it is a great relief :) It says the module is 92% full. The firmware you provided the link for is the one I used originally. I tried to upload it and update via this upload page but I get the message "Update Cancelled". Appreciate your further guidance.
KL

[image]
[image]

KL-XL, 2020-06-21:
OK. It works now :) After 5 repetitions (firmware flashing, file uploading, page refreshing) the setup wizard kicked in and I was able to access the ESP3D GUI. Thank you very much oldman4U for your assistance. 
Regards,
KL

closed 2020-06-21 by KL-XL