bigtreetech/BIGTREETECH-TFT35-V3.0 #68: SD Card Folder Names in TFT Mode

KL-XL, 2020-06-18:
Hi there,

When browsing the onboard SD card in TFT mode the folder names are shown as 8.3 filenames (like MS-DOS and alike) : PRINTA~1, PRINTP~1, 47ACB~1.MOD, etc. This doesn't happen in the Marlin mode where the same files look like; Print ABS, Print PETG, Print PLA, etc. Is there any configuration setting that can be changed in the firmware or in the UI please?

Thanks,
KL

oldman4U, 2020-06-20:
Hi KL-XL.

There are Long Filename support functions in the TFT firmware (config.ini)and in the configuration_adv.h firmware of the mainboard which have to be turned on. Check the settings and let me know then.

KL-XL, 2020-06-20:
Many thanks oldman4U! Is there a way to set this property via G-Code in pronterface please?
Thanks again

oldman4U, 2020-06-21:
Hi.

You are very welcome.

Long file and folder names are an issue, especially when using a touchscreen and i would recommend to use a mx. of 10 digits.

I do not believe that this can be set using code commands, but Pronterface is not my world, sorry. Maybe you would like to start a new ticket in the general BTT Touchscreen repository, which is well covered by many users.

Slightly off topic...
When i started with this hobby almost a year ago, i had no idea what VSC, Marlin, Compile and so on really means. But it was my feeling, that if i would like to get everything out of the hardware which is possible, i would have to learn all those things - and i was right. It is really funny to change parameters and suddenly the printer works different or can do new things. It will take a day or tow to get used to it, but i highly recommend every user to make this step.

Please let me know in case there is anything else i can help with. In case you do not need a ticket, please close it.

Thank you

KL-XL, 2020-06-21:
Thanks again oldman4U. Looks like I will need to find some time to get into the code. I am a developer in VB and C# DOTNET, so it shouldn't be an issue :) My concern is the risk to damage the hardware :). I will leave this open for a week or see in case someone has additional thoughts on the possibility of changing the setting via G-Code console.

oldman4U, 2020-06-22:
You are very welcome. Glad i could help.

Not sure there is a way to harm the hardware by changing the firmware, but to have a working firmware by the hand as a fallback just in case, is always a good idea.

Happy printing

KL-XL, 2020-06-22:
> Not sure there is a way to harm the hardware by changing the firmware,

Hahaha, try connecting BLTouch to the Z-Probe pins on BTT SKR Mini E3 V2.0, install the firmware "BLTouch with Z-homing",  press the home button and watch the crash-landing (DON'T DO IT :))

oldman4U, 2020-06-22:
But it is still the hardware which does the crash;-) Sure there are
possible combinations which should not work but in case hardware breaks
everything is possible.

With your skills you are a master within 2 days.

;-)

oldman4U, 2020-07-02:
Hi KL-XL.

Could you please close this ticket, thank you.

Hope you found some time to check the firmware in the meantime;-)

closed 2020-07-02 by KL-XL