bigtreetech/BIGTREETECH-TFT35-V3.0 #69: Tft35 e3 unable to find printer ..ender5 creality silent board!!

sinbad169, 2020-07-01:
hi can anyone help, I've seen videos of ender3 screen swop and they work, apart from the unable to find printer error, I have changed the baud rate as suggested and this has made no difference on mine, the marlin emulator works fine so I know its talking to my printer,  I only have the lcd ribbon connector fitted ( as the videos seen have) I am running the creality silent board...is it me or are the videos not telling the whole story

oldman4U, 2020-07-01:
Hi sinbad169.

For the touchscreen to work, a second cable has to be connected called the serial connection. With the cable you use, ONLY the Marlin mode works.

Hope this answers your question

sinbad169, 2020-07-01:
Thank you oldman4U.
Just ordered a  skr mini e3 v2 so hopefully thats the cure .

oldman4U, 2020-07-02:
Hi.

I am sure this will work. Could you please be so kind and close this ticket, thank you.

You can start a new ticket whenever you want.

closed 2020-07-02 by sinbad169