bigtreetech/BIGTREETECH-TFT35-V3.0 #70: TFT35-E3 V3.0 Illegal FLASH APP! error on firmware upgrade

aztrip, 2020-07-04:
NEWB ALERT: Just installed  SKR Mini E3 v2.0 with TFT35-E3. Everything seemed to work well and it was recommended to update the firmware to the latest version. Following the instructions; copy .bin and TFT35 to rott of sd card. It appeared to update nicely and Removed power, removed sd card and powered up to ILLEGAL FLASH APP!

Any solutions out there?

oldman4U, 2020-07-04:
Hi aztrip.

This is lightly the wrong repository, but i will try to help.

Are you familiar with compiling your own firmware, or have you been using the precompiled firmware from this repository? Maybe you can send me a screenshot of the content of the SD card.

Thank you

aztrip, 2020-07-04:
Thank you for responding. Being a newbie to the TFT display and to GitHub I just guessed. Anyways I discovered the problem was I used an as card that was formatted as FAT16. Once I reformatted to FAT32, the problem was resolved. Thanks again for the offer.

oldman4U, 2020-07-04:
Great to hear you could solve the problem.

Please be so kind to close this ticket.

Thank you

closed 2020-07-04 by aztrip