bigtreetech/BIGTREETECH-TFT35-V3.0 #71: TFT35 displayed whitescreen after changing the knob LED color to white from off

Dannydeleon0308, 2020-07-10:
I just got my TFT35-V3-E3 with firmware version 26.xx and when I first plugged it in, it was working fine until I changed the knob LED color from none to white. It then just displayed a white screen. I tried pressing the reset button but still it didn't worked. The white LED presisted even between power cycling the printer and I have no way of turning the knob LED off. Is there a way to turn the LED off outside of the UI? I tried reflashing it with the latet firmware on the github site but the LED light is still on and it is still displaying a white screen.

Dannydeleon0308, 2020-07-10:
I forgot to mention that I am using a atMega 2560 with a Ranps 1.4 board. I currently have a reprap discount full graphics display which works flawlwssly.  the TFT35 is very inconsistent, sometimes it works and when you power cycle the printer, it doesn't work anymore. And BTW, this is my second one, I returned the first one thinking it was defective but the new one is behaving exactly the  same.  I am wondering if my ramps board is supplying enough voltage to the display? Any ideas on what to check? Thanks.

oldman4U, 2020-07-10:
Hi. This is the repository for the non E3 TFT but maybe i can help. 

To reset the TFT's touch screen calibration, create a blank file named reset.txt and place in root folder of the sd card. Insert the SD card into the TFT's SD card reader and power cycle your printer to start the reset process.

Unfortunately I can not say if this is mainboard related but my screen is the same and does not show this behavior.

oldman4U, 2020-07-11:
Hi. Any news regarding the reported issue ?

Dannydeleon0308, 2020-07-11:
HI,

Thanks for your help. I did the reset.txt file and it did turned off the LED of the knob but I am still having the inconsistent behavior as before. Sometimes it works and after a power cycle it will not work. I have tested the voltage being sent to the Serial port of the TFT35 from my ramps board and it is indeed 5V. I don't know what is wrong but I will keep it for now. Maybe I will change my control board to a 32 bit one from BTT and see if it works then. Thanks again.

oldman4U, 2020-07-12:
Hi. Could you please remove the display from the metal frame it is installed in and see if this solves the remaining issues.

oldman4U, 2020-07-22:
Please close this ticket in case the problem is solved. If not, please let me know. 

Thank you

Dannydeleon0308, 2020-07-31:
I am closing this issue as I decided to just upgrade my control board from a ramps to an SKR 1.4 turbo which works well with the TFT35. Thanks for your help.

closed 2020-07-31 by Dannydeleon0308