bigtreetech/BIGTREETECH-TFT35-V3.0 #72: Print freezes when changing speed or flow

Whiskaya, 2020-07-20:
I've had several instances when, while tuning speed or flow, suddenly the numbers go haywire and get scrambled (ONLY the flow/speed numbers), jumping randomly.
A few seconds later, the nozzle stops and stays hot, but doesn't move anymore.
Can't pause/resume, only stop print and start everything from the top.

Ender 5
TFT35 V3.0
SKR Mini E3 V2.0
Stock PSU

oldman4U, 2020-07-21:
Hi.

This issue is fixed in the latest version of the TFT firmware.

Please try and let us know if you ned any help

Whiskaya, 2020-07-21:
It came with V3.0.26 from factory, and already had some configuration changes like Marlin mode not showing the title bar.

Still, had some problems/bugs, so I downloaded fresh firmware from Github. It was the same V3.0.26. This was 2 weeks ago.

Today I noticed the firmware was re-uploaded 8 days ago. Is that file corrected, even though it is the same version number?

oldman4U, 2020-07-21:
Forget the version number, use the v26 from today and enjoy even the new
image preview when using Cura as the slicer.

It is great and your problem will also be fixed.

Happy printing

Whiskaya, 2020-07-21:
Indeed, it's working fine so far. Thanks!

closed 2020-07-21 by Whiskaya

oldman4U, 2020-07-22:
Also for me.

👍🏻