bigtreetech/BIGTREETECH-TFT35-V3.0 #74: BTT TFT35-V3.0 Screen Flashes

Ripper80, 2020-08-06:
I just upgraded my Ender 3 Pro to SKR 1.4 Turbo (firmware not installed yet) and the TFT35.  When I power up the system the magic smoke did not escape, but it remains blanks and flashes....waited for about 30 seconds and then powered off times 2.  Is this normal?  I heard it was fairly quick, don't want to take a chance and break it...I have both EXP cables plugged in, and confirmed they are correctly installed and the 232 cable.

The EXP cables are NOT the ones that came with the touchscreen as the cables were too short (moved everything to the back of the printer).

Any advise/ guidance would be appreciated.  THANK YOU

oldman4U, 2020-08-07:
Sure you need 2 EXP cables and not one?

oldman4U, 2020-08-08:
To connect a SKR board a single exp cable must be connected to exp 3 of the tft. Please change and let me know.

Ripper80, 2020-08-12:
Any news on what? I asked a question and this is the only thing I have seen as a response....not sure if my email misidentified as spam, but will check shortly...

⁣Get BlueMail for Android ​

MaasonR, 2020-08-12:
Ok, now I see the replies. The instructions I have seen indicate that to use native Marlin screen and the BTT touchscreen, need the TFT cable connected to RS232 AND the two EXP cables.  I flashed the TFT a second time and changed the baud rate to 250000 and updated MARLIN to same baud rate and same issue. Videos said to switch so reset switch on TFT is on left and that is how it was set, switched it (upside down) and reset on right, everything came up.  Now the only issue I have is that it shows NO PRINTER ATTACHED. Currently, my entire motherboard is once again completely disconnected to determine what I have miswired or not for a good connection on.  BUT the screen works as far as I can tell

oldman4U, 2020-08-12:
Please show me why you believe that you have to connect 2 EXP cables to the SKR board.

MaasonR, 2020-08-12:
Being new to all things 3D, I rely on instructions from people I have been referred to to help with most things 3D printing...  examples are these two:

https://youtu.be/DodRRgZf5Lo

https://youtu.be/l1Putn10_Ek

Lastly, the GITHUB documents (https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/blob/master/Hardware/Explanation%20of%20black%20screen%20in%20TFT35%20V3.0.pdf) state that EXP 1 and EXP2 are supposed to be connected.  

All three state connecting the three cables.  Other than your comment, which I didn’t see until this morning and HAVE NOT TRIED, I have seen nothing in any form stating what you idea.  I will try later today after work.  I have NOT dismissed your comment, just letting you know what I had tried in response to your question “Any news?”.

BobP2020, 2020-08-12:
> 
> 
> Please show me why you believe that you have to connect 2 EXP cables to the SKR board.

When using a BTT TFT35 display with a BTT SKR1.4 control board and you want both "Marlin Mode" and BTT's touch screen mode the 3 cables are needed. HOWEVER on several printers, if you are using the printer's "stock display" only one cable is needed for the "Marlin Mode". Well, this is my understanding.

oldman4U, 2020-08-12:
I am new to this also😁

I use the TFT35 with an SKR E3 DIP and with such a board only one EXP is used. But reading the SKR v1.3/4 manual I would also connect two cables and you wrote the screen works so far. The no printer connected means that the TFT can not see the mainboard via the 232 serial cable. So disconnecting this cable should give you the same result.

oldman4U, 2020-08-18:
Hi

Any news?

Ripper80, 2020-08-20:
Flashing screen is gone...will close this...the NO PRINTER ATTACHED that currently plagues me is still in effect.

closed 2020-08-20 by Ripper80

Mayoogh, 2021-07-24:
> Flashing screen is gone...will close this...the NO PRINTER ATTACHED that currently plagues me is still in effect.

How did you fix the Flashing screen?

diyoyo, 2022-02-25:
Hi, 
i don't know whether you fixed you "No Printer Attached" Problem. But here is what worked for me.
I had the same flickering screen problem + a dimming problem where the screen brightness just converged to 0 over a few seconds.
On the back of the TFT, there was 2 slots for a 5-pin connectors, and the flickering screen was only happening when I had followed online tutorials telling to plug to the RS232.
In fact, when I was pluging it to the SWD slot, the screen was not flashing/flickering/dimming anymore, and worked perfectly fine. Maybe that's what you did too?

But then, I had the "No Printer Attached", which did not happen when plugging to the RS232 slot.

So I decided to do a mix of both: I plugged 4 cables into the RS232 slot, removing the 5V pin.
Then I plugged the fifth cable to the SWD slot, in the 3.3V pin.

And now it works and also detects the printer.

I am facing the next challenge, which is to solve the "bed levelling error" that the screen now throws at me.

Hope that helped someone.

Disclaimer: I don't know anything about electronics. Maybe this will fry my board, I have no clue. Try it at your own risks.
My guess here was that it was a voltage problem.