bigtreetech/BIGTREETECH-TFT35-V3.0 #75: TFT-35 E3 V3.0 goes blank after 5 seconds of turn on of an Ender 3 pro with BIGTREETECH SKR Mini E3 V2.0

chobson303, 2020-08-14:
I'm trying to upgrade my Ender 3Pro with the SKR Mini E3 V2.0 and the TFT-35 E3 V3.0 touch Screen.  After installation I turn on the printer and the screen comes on for about 5 seconds and then goes blank.  How can I fix this?

oldman4U, 2020-08-14:
Pictures which show the status you describe?

Have you tried to remove the display from the metal frame it is connected to to see if this solves the problem?

chobson303, 2020-08-15:
The display is only connected to the SKR mini board via the cables only, it's not touching any part of the printer and it still shuts off within 5 seconds.

oldman4U, 2020-08-15:
What is shown in the 5 sec and after that? Are you using the original cables? What if you use the original display?

oldman4U, 2020-08-15:
And has it working before  Have you compiled your own firmware. What happens when you upload new firmware to your TFT.

chobson303, 2020-08-16:
Here’s a link to my Youtube video showing the problem.https://youtu.be/l10GX6TK3SU ThanksCharles Sent from Mail for Windows 10 From: oldman4USent: Saturday, August 15, 2020 1:49 AMTo: bigtreetech/BIGTREETECH-TFT35-V3.0Cc: chobson303; AuthorSubject: Re: [bigtreetech/BIGTREETECH-TFT35-V3.0] TFT-35 E3 V3.0 goes blank after 5 seconds of turn on of an Ender 3 pro with BIGTREETECH SKR Mini E3 V2.0 (#75) And has it working before Have you compiled your own firmware. What happens when you upload new firmware to your TFT.

oldman4U, 2020-08-16:
Ok. Would you be so kind to answer my questions above. 

Maybe you can attach you config.ini file here also

chobson303, 2020-08-16:
The new touchscreen was not attached to the medal frame in the video.  I am using the new cables that came with the kit.  This problem existed from the initial installation.  I just replace the old motherboard and old screen with the new ones, and the first time I turned on the unit which was at that time attached to the medal frame, all I got is just the 5 seconds of screen and nothing else.  I removed the medal frame and placed the screen on the desk only attached to the cables and the same thing, 5 seconds of screen then blank.  I put the original screen back on and turned on the unit.  The screen does come on but very lightly and then it goes blank too after about 5 seconds. I have not tried to install new firmware. I don’t know how to get the config.ini file to send you but I have attached is a copy of the firmware file that was on the micro SD card that came with the display.  It also had the SKR-Mini-E3-V20 file on the card as well.Thanks  Sent from Mail for Windows 10 From: oldman4USent: Sunday, August 16, 2020 9:01 AMTo: bigtreetech/BIGTREETECH-TFT35-V3.0Cc: chobson303; AuthorSubject: Re: [bigtreetech/BIGTREETECH-TFT35-V3.0] TFT-35 E3 V3.0 goes blank after 5 seconds of turn on of an Ender 3 pro with BIGTREETECH SKR Mini E3 V2.0 (#75) Ok. Would you be so kind to answer my questions above.Maybe you can attach you config.ini file here also

oldman4U, 2020-08-18:
Hi. Sounds like a hardware issue to me. But could you remove the rs-232 serial cable and try it again.

chobson303, 2020-08-19:
I unplugged my BLTouch and now the screen works.  But the screen says no printer attached and now how do I get my BLTouch to work? Sent from Mail for Windows 10 From: oldman4USent: Tuesday, August 18, 2020 2:29 PMTo: bigtreetech/BIGTREETECH-TFT35-V3.0Cc: chobson303; AuthorSubject: Re: [bigtreetech/BIGTREETECH-TFT35-V3.0] TFT-35 E3 V3.0 goes blank after 5 seconds of turn on of an Ender 3 pro with BIGTREETECH SKR Mini E3 V2.0 (#75) Hi. Sounds like a hardware issue to me. But could you remove the rs-232 serial cable and try it again.

chobson303, 2020-08-19:
I can’t seem to find the user guide for the SKR Mini and the touch screen.  Can you help direct me to where they are? Sent from Mail for Windows 10 From: oldman4USent: Tuesday, August 18, 2020 2:29 PMTo: bigtreetech/BIGTREETECH-TFT35-V3.0Cc: chobson303; AuthorSubject: Re: [bigtreetech/BIGTREETECH-TFT35-V3.0] TFT-35 E3 V3.0 goes blank after 5 seconds of turn on of an Ender 3 pro with BIGTREETECH SKR Mini E3 V2.0 (#75) Hi. Sounds like a hardware issue to me. But could you remove the rs-232 serial cable and try it again.

oldman4U, 2020-08-19:
Where have you connected the BLTouch?

Unfortunately I can not help with this mainboard. I use DIP and PRO but not mini.

chobson303, 2020-08-24:
I sent it back and ordered another one.

oldman4U, 2020-08-24:
Makes sense.

Please do not forget to close the ticket once you do not need it anymore. Thank you

chobson303, 2020-08-24:
The replacement V1.2 main board and the Touchscreen seems to be working fine.  I'll do a final update once I get a good print.  But it appears something was bad with the V2 mainboard and TFT-35 E3 v3 touchscreen.

oldman4U, 2020-08-25:
Good to hear that it works. I always have some spare parts and sometimes they are not the same so i can try different things and play around.

oldman4U, 2020-08-30:
Hi. Please close this ticket. Thank you

chobson303, 2020-08-30:
Done

closed 2020-08-30 by chobson303

cyburrs, 2021-02-27:
I have the exact same problem. I bought both the TFT 3.5 V3 and the SKR Mini E3  V2.0.
Updated the firmware to Ender 5 with BLTouch.
Quits after 5 seconds. 
Removed the BLT touch and it works. Now how do I get my BL Touch to work?