bigtreetech/BIGTREETECH-TFT35-V3.0 #76: TFT35 Encoder knob sets speed no matter which screen you're looking at

leopshef, 2020-08-25:
The encoder knob changes the speed value when turned, meaning if you use the encoder knob to set your temperature value for example then the speed is also changed.

oldman4U, 2020-08-25:
Yes. And because of this BTT turned off the ability to use the knob while using the touch screen, but users complained so BTT turned it on again. I am not sure there is a good solution for this which works for all users.

leopshef, 2020-08-25:
It's fine for it to work, however it shouldn't change the speed when I'm using it to control what my temperature is.
If I'm on the set Hotend temperature screen, and I twist the encoder to change my temp to 240c, when I go back to the homescreen my speed is 200%. 
Is there no way to have the speed only update if the speed screen, or even speed/home screen is the one selected?

I'm not sure why people need to be able to change speed by default anyway tbh

leopshef, 2020-08-25:
Sorry, I know why they need to change speed, but I don't know why the encoder's default behaviour is controlling speed.

oldman4U, 2020-08-26:
It seems that there is no way to enable the knob without sending the pulses
also to Marlin, even the touchscreen is active. Check the closed tickets
related to this.

oldman4U, 2020-08-26:
In Marlins main window you control the speed with the knob. Because of this
you change the speed and not something else.

leopshef, 2020-08-26:
Ah that's a shame.
The encoder going to Marlin automatically controls speed then I assume.
I'll have a look around for other tickets related to this.

Thanks for getting back to me.

leopshef, 2020-08-26:
Ah right, I understand.

Thank you!

closed 2020-08-26 by leopshef

oldman4U, 2020-08-26:
You are welcome!

👍🏻