bigtreetech/BIGTREETECH-TFT35-V3.0 #77: SD init fail in BTT mode and No media in Marlin mode

fineobjetcs, 2020-08-26:
Hi All! I have troubles with my BTT 3.5 tft v3 and SKR 1/4T
in marlin mode i see SD insert and removed when i insert and remove SD card but see NO Media when i want to print from SD.
In BTT mode when i insert SD screen print SD int fail

Also in BTT mode i see U Disk with media but when i print from UDisk my printer start heating, go to Z0 and stop
in this case i see Bad LevelingON on screen and printing No start? newer

Whats wrong ?

oldman4U, 2020-08-27:
What is the size and format of the sd card?

Is it a new config or has this working before?

Attach a gcode file.

oldman4U, 2020-08-27:
BTW. It seems that you are collecting tickets. It is important that you close tickets which you do not need anymore. Otherwise other users will have problems finding tickets which already cover their question.

Please close https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/146

Thank you

oldman4U, 2020-08-30:
Please reply or close this ticket. Do you know how to close tickets?

fineobjetcs, 2020-09-02:
Hello !  Its new setup, i bought new skr+new screen
i try SD and USB/ format is Fat32 size from 128 mb to 4 gb/ i try many cards and many gcode files/

AnHardt, 2020-09-02:
And what was the result?

fineobjetcs, 2020-09-02:
No result ( i cant see SD or U-drive in Marlin mode, cant see SD in BTT mode
and when i start printing anything from u-drive in Btt mode printers stop after Bad LevelingON

fineobjetcs, 2020-09-02:
i take video for you and send it via Youtube link

fineobjetcs, 2020-09-02:
https://youtu.be/tPmVzpOob1Y

https://youtu.be/4tj61mzgmII

https://youtu.be/-blJyF2AKL0

https://youtu.be/Ct_k8v9FZOY

oldman4U, 2020-09-02:
Hello.

Thank you for closing the other ticket. I really appreciate.

Watching your movies i get the feeling, that there is a complete mismatch of software and configuration OR a hardware issue. Do you have any spare parts you could try? Can you also send a picture of the Info window which you can find in the Menu of the TFT. Hopefully i get my SKR 1.4 this week, so i can test your setup (even I have no delta, but printing from disk should be possible independent of the printer type;-))

fineobjetcs, 2020-09-03:
Iv got another SKR 1/4t for testing. I sure that it only software error but i cant find were (((

oldman4U, 2020-09-03:
So the new board has the same issue? Oryou have not tried it yet?

fineobjetcs, 2020-09-03:
Im not tried yet) i recived new board day ago _)

С уважением

Сергей Семенов
+7 (916) 671-1689

Руководитель студии
предметного дизайна fineobjects
www.fineobjects.ru

fineobjetcs, 2020-09-03:
> Ok.
>
> Please use theSDcard which comes with the board also for firmware update
> and printing and try very short filenames.
>
> Let me know

Ok i will try, but im sure that long filenames cant inhibit SD init

oldman4U, 2020-09-03:
I am not sure so it would be great if you could try it without.

Thank you

oldman4U, 2020-09-07:
Hi.

Any news?

fineobjetcs, 2020-09-07:
> I have no idea what i need to do

wait btt support

oldman4U, 2020-09-07:
So you got a new SKR board and you replaced the old one with the new one?

What is the result using short file names?

fineobjetcs, 2020-09-07:
Iv got new data ! )))
when i try to use   #define SDCARD_CONNECTION ONBOARD
 my 3.5TFT works in Marlin mode with SKR internal SD card and also
work in BTT mode with Screen SD Card and U-Drive
but in Marlin mode cant see screen SD card and in BTT mode cant see
SKR internal SD card
There is no difference with short and long names )

oldman4U, 2020-09-08:
As far as i know Marlin can not see the TFT at all and can use SKR board only. 

What the TFT can see you can define in config.ini.

oldman4U, 2020-09-17:
Have you checked the config.ini for sd support?

BobP2020, 2020-09-17:
With regard to the comment from above "but in Marlin mode cant see screen SD card"...correct. When the system is doing an LCD12864 Simulation, A.K.A. "Marlin Mode" the g-code files located on the SD card inserted into the SKR1.4 are the only files accessible for printing. (Just to give a thumbs-up to the comment by oldman4U).

oldman4U, 2020-09-17:
At leadt i have not found a way to do this and as far as i understand how the different modes communicate with the mainboard - the Marlin mode can not see the TFT and therefore the „storage attached to it“.

Marlin sees mainboard based storage only
Touchmode sees TFT Udisk and sd card and if configured to do so, also the mainboards sd card.

oldman4U, 2020-09-20:
@BobP2020 we are both wrong🥺

If the SD card of the TFT is usable in Marlin mode or not depends on the type of connection to the mainboard. Using the CR10 style connection with a single EXP connection, the SD card of the TFT can not be used with Marlin emulator. 

Using a RepRap Full Graphis TFT which connects with two EXP cables to the mainboard, it is hardware wise possible to use the SD card of the TFT in Marlin emulation mode. In this case you can define in Marlin which SD card you want to use, from the Board or LCD.

oldman4U, 2020-09-20:
@fineobjetcs 

With your hardware you should be able to get the following options to print from (besides from USB and WIFI). 

In Marlin mode:
SD card from mainboard **OR** SD card from TFT.

In Touchscreen mode:
You can print from U disk and the SD card you defined in Marlin.

BobP2020, 2020-09-20:
Well, OK on that. I was thinking along the line of having the same feature
that you have when operating in touch screen mode. That is, you have “both
“ SD cards available. As a matter of fact when in touch mode you have 3
possible choices from which to print. Either one of the SD cards or the
USB. The best of all worlds 😃.

oldman4U, 2020-09-20:
I agree. We print only from Touchscreen using U disk even we also used UsB and WIFI and sd card before.

But that it is possible to use the TFTs SD card in Marlin i learned today and so I wanted to share this....

oldman4U, 2020-09-23:
Any news?

Please let us know