bigtreetech/BIGTREETECH-TFT35-V3.0 #78: Screen not lighting up

xxKeoxx, 2020-08-31:
I’m trying to install this on an SKR 1.4 turbo on my Ender 5 plus.  I saw the screen power on but then it slowly faded to black. I can hear the touch screen respond when I press it, but I cant see what I am pressing. 

Any advice on how to fix this?

oldman4U, 2020-08-31:
Replace the broken TFT.

xxKeoxx, 2020-08-31:
> Replace the broken TFT.

The TFT works fine on another printer.  The board is fine because the printer prints fine using octoprint. It’s almost like the screen isn’t getting enough power.

oldman4U, 2020-09-01:
Ok.

Then try another 5v source. It has ti lit up with EXP 1 or RS232 connected
only.

bsg6867, 2020-09-01:
Hi there, I have the same issue going on. I've hooked the TFT35-E3 V3.0 to the SKR V1.4 Turbo with all 3 wire hookups.

When reading from the TFT connection on the mainboard I'm receiving 2.65V out of the 5V end, same with the EXP ports. Could that be the issue? If so, any ways to fix that?

xxKeoxx, 2020-09-01:
> Hi there, I have the same issue going on. I've hooked the TFT35-E3 V3.0 to the SKR V1.4 Turbo with all 3 wire hookups.
> 
> When reading from the TFT connection on the mainboard I'm receiving 2.65V out of the 5V end, same with the EXP ports. Could that be the issue? If so, any ways to fix that?

I think I am getting closer to finding my issue.  In my case the Ender 5 + must have had a different pin layout, because my BLTOUCH had it's power line on the outside not the middle like the board expected.  I had to re-arrange my wires in the plug so red was in the middle.  Now I have power to both devices, and I am working through the firmware.

oldman4U, 2020-09-02:
@xxKeoxx Sounds good to me. Thumb up! Please close this ticket once you do not need it anymore. Thank you

@bsg6867 Disconnect everything you do not need as a minimum to test the display (so finally all you need is 14/24V power) and try it again. Could be broken hardware also, not only the TFT but also the power supply or a fuse on the mainboard.

bsg6867, 2020-09-02:
@oldman4U I actually just got it figured out this morning. I had 2 of my wires on the BL Touch mount backwards. Which apparently was causing it to draw too much, and that's why my display wasn't working. 
Now to get the BL Touch to work and I'll be all set!

closed 2020-09-02 by xxKeoxx

oldman4U, 2020-09-03:
Two times the same reason. Cool👍🏻